qiboteam/qibocal #100: readthedocs documentation

scarrazza, 2022-10-11:
In order to generate documentation for readthedocs, qibolab must be public, I had just setup the building process but it crashes due to access to qibolab code repo.

On the other hand, I believe at some point qibolab could be a simple dependency.
@andrea-pasquale and @stavros11 any thoughts?

andrea-pasquale, 2022-10-11:
I had a similar issue when try to use `dependabot` to update some deps.
As soon as we make qibolab public everything should be easier.

andrea-pasquale, 2022-10-21:
@scarrazza since now qibolab is public could you please generate the documentation for readthedocs?

scarrazza, 2022-11-01:
@andrea-pasquale, I believe we are missing some dependencies, see [this build](https://readthedocs.org/projects/qibocal/builds/18502569/).

andrea-pasquale, 2022-11-01:
Yes, this is happening because we have updated the deps in qibolab. 
I've already a PR in order to fix this (#135). I believe that if we merge it to main also readthedocs should compile without issues.
The PR should be ready to merge to main.

scarrazza, 2022-11-01:
We are close: https://readthedocs.org/projects/qibocal/builds/18505459/

andrea-pasquale, 2022-11-01:
This is strange...
I'll have a look.

andrea-pasquale, 2022-11-01:
There was a typo the correct module is `sphinx_markdown_tables` and not `sphinx-markdown-tables`.
I've pushed to main the pyproject.toml with the correct module.

scarrazza, 2022-11-01:
Not yet: https://readthedocs.org/projects/qibocal/builds/18506595/

andrea-pasquale, 2022-11-02:
> Not yet: https://readthedocs.org/projects/qibocal/builds/18506595/

@scarrazza can I have access to this page to debug?

scarrazza, 2022-11-02:
In principle all builds are publicly visible: https://readthedocs.org/projects/qibocal/builds/
The build starts when you commit to main.

andrea-pasquale, 2022-11-02:
I see, I thought that you can trigger manually when to compile the documentation, my bad.

scarrazza, 2022-11-02:
Register to the website and send me your email username by email, I can try to link so you can trigger manually.

andrea-pasquale, 2022-11-02:
Readthedocs is finally working.
I'll close this issue.

closed 2022-11-02 by andrea-pasquale