qiboteam/qibocal #101: Frequency Estimators

Edoardo-Pedicillo, 2022-10-11:
# Introduction

I would like to propose and compare some methods for finding the peak given a set of uniformly sampled, non-repeating frequency-MSR pairs coming from resonator spectroscopy.

The data for this analysis comes from resonator spectroscopy with 80 frequency-MSR couples, which can be seen in the following plot.
[image]

# Model fit  

We fit the data with a combination of a Lorentzian and a linear model, using the ` lmfit`  library.  The peak of the Lorentzian model is considered the “true value” of the peak and it is used as benchmark for the following algorithms.

[image]

# Methods
## 1. Fit 

The peak is the Lorentzian's center of the Lorentzian+linear model. 

``` python
from lmfit.models import LorentzianModel,LinearModel

def Lorentzian_fit (X,Y):
    
    lor_mod = LorentzianModel(prefix = 'lor')

    pars = lor_mod.guess(Y, x=X)

    # linear model
    lin_mod = LinearModel(prefix = 'lin')

    pars.update(lin_mod.make_params(intercept=Y.max(), slope=0))

    # the used model is the sum of the two models 

    mod = lor_mod + lin_mod
    out = mod.fit(Y, pars, x=X)
    
    return out.best_values

def fit_model (msr,freq):
    parameters = Lorentzian_fit (np.array(freq),np.array(msr))
    return parameters['lorcenter']

```

## 2. Min

Given a set of frequency-MSR couples, the peak corresponds to the frequency with the lowest MSR value. It is implemented as benchmark.

``` python
def min_method (msr,freq):
    
    return freq[msr.index(min(msr))]
```

## 3. Quadratic method 

Looking at the picture below, the main idea of this method and the following ones is to find the $m_k$  value, between the sampled data, that maximizes the function $X(m)$, and evaluate the position of the peak ( $m_{peak}$ ) with a formula depending on the nearest values ( $m_{k+1}$ and $m_{k-1}$).

[image]

``` python
def quadratic_method(msr,freq):
   
    msr = [1-i for i in msr]
    k = msr.index(max(msr))
    if k<len(msr)-1:
        delta = (msr[k+1]-msr[k-1])/(2*msr[k]-msr[k-1]-msr[k+1])
        peak = freq[k]+delta
        
    else:
        peak = msr[k]
    
    return peak

```

## 4. Quinn's method 

``` python
def quinn_method(msr,freq):
    # freq must be sorted
    msr=[1-i for i in msr]
    k = msr.index(max(msr))
   
    if k<len(msr)-2:
     
        alpha_1 = msr[k-1]/msr[k]
        alpha_2 = msr[k+1]/msr[k]
        
        if alpha_1 == 1:
            return (freq[k-1]+freq[k])/2
            
        if alpha_2 == 1:
            return (freq[k+1]+freq[k])/2
            
        delta_1 = alpha_1/(1-alpha_1)
       
        delta_2 = -1*alpha_2/(1-alpha_2)
    
        if delta_1>0 and delta_2>0:
            
            delta = delta_2
            
        else:
            
            delta = delta_1

        peak = freq[k]+delta
        
    else:
        peak = msr[k]
    
    return peak
```

## 5. Grandke's method

``` python
def grandke_method (msr,freq):
    # freq must be sorted
    msr=[1-i for i in msr]
    k = msr.index(max(msr))
    
    if k < len(msr)-1:
        
        msr_max_neighbor = max(msr[k-1],msr[k+1])
        alpha = abs(msr_max_neighbor / msr[k])
        delta = (2*alpha - 1) / (alpha + 1)

        peak = freq[k]-delta
        
    else: 
        peak = freq[k]
        
    return peak 
```
## 6. Macleod method 

``` python
def macleod_method (msr,freq):
    
    # freq must be sorted
    msr = [1-i for i in msr]
    k = msr.index(max(msr))
    if k<len(msr)-1:
        delta = (msr[k+1]-msr[k-1])/(4*msr[k]-2*msr[k-1]-2*msr[k+1])
        peak = freq[k]+delta
        
    else:
        peak = msr[k]
    
    return peak

```

# Comparison

For each algorithm, the peak is evaluated several times with different data, sampled from the initial dataset, containing a different number of elements. Performance is evaluated in terms of the mean of the peaks, the standard deviation and the MSE with the 'true value' of the peak.

The fit method is the worst one, and the other ones assure better results than the min method when the number of samples is low ( 20 samples circa)

[image]

Let me know if you like this approach or have any suggestions. @scarrazza 

# References 

[1] E. Jacobsen and P. Kootsookos, "Fast, Accurate Frequency Estimators [DSP Tips & Tricks]," in IEEE Signal Processing Magazine, vol. 24, no. 3, pp. 123-125, May 2007, doi: 10.1109/MSP.2007.361611.

[2] https://www.researchgate.net/publication/253597963_On_Local_Interpolation_of_DFT_Outputs

[3] B. G. Quinn, "Estimating frequency by interpolation using Fourier coefficients," in IEEE Transactions on Signal Processing, vol. 42, no. 5, pp. 1264-1268, May 1994, doi: 10.1109/78.295186.

scarrazza, 2023-01-23:
@Edoardo-Pedicillo could you please check with @igres26 if this issue can be closed?

AleCandido, 2023-03-07:
Actually, since they are very simple, I would implement all of them in Qibocal, and test on a larger amount of data.

E.g. in this the `min` works good enough, but we have seen more cases in which the background would hide the peak.
Moreover, it would be interesting to be able to retrieve also multiple peaks from a single scan (since it might happen).

andrea-pasquale, 2023-12-18:
Given that we are moving towards a NN approach I think we can close it.

closed 2023-12-18 by andrea-pasquale