qiboteam/qibocal #103: Completing documentation

scarrazza, 2022-10-11:
In order to provide an useful documentation please find below my wishlist (be free to include more features):
- [ ] complete installation instructions. Include CI tests for linux, mac and windows for multiple version of python. So we can tell which configurations are supported.
- [ ] complete qq-upload docs
- [ ] include references to real actions names, cleanup examples, include links to examples. Add step by step instructions explaining how to run the code.
- [ ] complete all calibrations by adding proper docstrings with explanation and references and including examples. If possible include some example of plots so people are aware of what expect. Someone new to the topic should understand what each action is doing.
- [ ] add documentation about live plotting, with some examples
- [ ] talk about the output data structure, show examples

scarrazza, 2022-10-13:
@andrea-pasquale concerning tests, if you want I can enable the qpu runner in this repo too. Let me know.

andrea-pasquale, 2022-10-13:
Thanks @scarrazza that would be great.

scarrazza, 2022-10-13:
Done.

PR #121 (2023-01-02): New documentation for qibocal
Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
- [x] Update `qq` documentation after merging #183

This PR will close #103.
I've started to refactor the documentation for qibocal. I'll try to cover all the points raised in #103.
If there are some issues with qibocal, or if we think that we are missing something in the documentation feel free to comment on this PR or in #103.
closed 2023-01-02 by scarrazza