qiboteam/qibocal #124: Request for being able to return multiple figures in a plot function

aorgazf, 2022-10-21:
Hi all,
While working on the cryoscope I have been wanting to display several plots: on the left, the phase and the derivative of the phase as a function of the flux pulse amplitude and duration; on the right, the cross sections for a particular flux pulse amplitude.
In order to do so I tried adding more subplots to the current figure, but the results are squeezed vertically, since the space allocated in the web page is constant. I guess it is possible, but I wasn't able to change the vertical size and location of the colorbar so that they are adjacent to its plot.
[image]

@andrea-pasquale would it be possible to change the interface so that the plotting functions can return more than one figure? perhaps a list, perhaps a custom collection of figures in which we could return additional information about how we would like them displayed. 
This is less important, but I am thinking that one may want to see additional plots that we may not want to show by default, for example the pulse sequence, the final position of the qubits in the blochsphere, polar plots of the evolution of the quadrature components of the received pulse, etc. The interface could provide either tabs or checkboxes to alternate between plots or to show/hide them.

maxhant, 2022-10-21:
One could have done it with multiple decorators, but it is not the best because the data is being processed from one plot to the next. It requires extensive calculations that one wants to do once. 

Note that subplots are not convenient because they share legends/ colourbars.

stavros11, 2022-10-21:
Thanks for opening this.

> In order to do so I tried adding more subplots to the current figure, but the results are squeezed vertically, since the space allocated in the web page is constant. I guess it is possible, but I [...]

You are right about the fixed space, I also don't like this but I could not fix it when doing the webpage template. In terms of code, what you need is in [line 165 of the html template](https://github.com/qiboteam/qibocal/blob/b652407722339539749780826932aca6131d2648/src/qibocal/web/templates/template.html#L165). Now the height is fixed to 500px. You can increase the allocated space by changing this. I think it's hard to find a value that works well for all kinds of plots but maybe it can help you to quickly get something that looks better in your example.

> @andrea-pasquale would it be possible to change the interface so that the plotting functions can return more than one figure? perhaps a list, perhaps a custom collection of figures in which we could return additional information about how we would like them displayed.

Returing a list of figures is possible, it just requires a few modifications to the template but it's not hard. The customization may be harder. My original idea was for the plotting functions to return blocks of html. This would be highly customizable, but unfortunately I was not able to make it work with the current way of using dash through iframe. @andrea-pasquale @scarrazza let us know your opinion about this.

> One could have done it with multiple decorators, but it is not the best because the data is being processed from one plot to the next. It requires extensive calculations that one wants to do once.

If the issue with this is code repetition, you can define a function that does the pre-processing and call it from all the different plot functions. If the issue is that the pre-processing is computationally heavy and you don't want to do it multiple times for the same data (which I agree is not good practice), there may be a work around using memoization (for example [lru_cache](https://docs.python.org/3/library/functools.html#functools.lru_cache)) but I have not tested it so I cannot say how well it would work in this example. These proposals may be useful if you want to implement something before we change the interface.

> Note that subplots are not convenient because they share legends/ colourbars.

I don't know much about plotly, I believe they should provide ways to customize these things, including not sharing colorbars, etc. but this will require a lot of time searching, reading their docs and trial and error which is a bit annoying.

andrea-pasquale, 2022-10-21:
Thanks for opening this issue @aorgazf

> Returing a list of figures is possible, it just requires a few modifications to the template but it's not hard. The customization may be harder. My original idea was for the plotting functions to [...]

This will surely be a nice idea if we can make this happen, or just simply be able to make the `height` parameter adaptive. I'm not an expert in html programming so I don't know if it is feasible or not. 

> One could have done it with multiple decorators, but it is not the best because the data is being processed from one plot to the next. It requires extensive calculations that one wants to do once.

Using multiple decorators is surely a valid solution for the time being. Regarding the "extensive calculations" as @stavros11 said we can have pre-processing function to load all the data for a specific routine once.

scarrazza, 2023-01-23:
Shall we close this issue?

andrea-pasquale, 2023-01-23:
There is a recent PR where @DavidSarlle is implementing this. I'll link the PR to this issue.

closed 2023-03-20 by aorgazf