qiboteam/qibocal #126: T1 fit seems to fail for long scans

aorgazf, 2022-10-21:
Hi @DavidSarlle,
T1 seem to fail when doing a long scan:

[image]

Could you please have a look into it?

DavidSarlle, 2022-10-23:
@aorgazf the problem is not the fitting. Bassically what is happening is if you set  the T1 delay_before_readout_end >= platform.repettion_duration, the routine crash because qblox driver is checkig for this assertion in /qibolab/src/qibolab/instruments/qblox.py (line 712):
                    time_between_repetitions = repetition_duration - sequence_total_duration
                    assert time_between_repetitions > 0

And this is not true for with the actual runcard where the "repetition_duration" is equal to 200_000 and the "delay_before_readout_end" = 200_000. If you increase the repetition duration in the runcard you can run larger T1 without problems.

In your case the failing fail was related with the same problem.

@aorgazf shall we increase teh repetition duration in teh runcard?

DavidSarlle, 2022-10-23:
@aorgazf as you can see in the following test, increasing the repetition duration accordingly to the T1 experiment, the fit and the function have no problems. 

[image]

aorgazf, 2022-10-24:
> @aorgazf shall we increase teh repetition duration in the runcard?

Thanks @DavidSarlle ,
In the runcard, no, at least for now, because `repetition_duration` directly impacts on the speed of execution of all routines.
If anything, we could change that value only in T1 routine if we saw that `delay_before_readout_end` is larger than it.
As part of readout optimisation we should determine the best `repetition_duration`, this is in our roadmap.

scarrazza, 2023-01-23:
Shall we close this issue?

closed 2023-05-31 by aorgazf