qiboteam/qibocal #132: poetry install fails linux 27/10/2022

aorgazf, 2022-10-27:
Hi @andrea-pasquale 
I've tried creating a fresh new environment in Saadiyat and `poetry install` fails:
``` bash
conda create -n qiboteam python=3.9
conda activate qiboteam
git clone https://github.com/qiboteam/qibocal.git
pip install poetry
pip install pre-commit
pip install pytest
pip install pytest-cov
clear
cd qibocal
poetry install
```

Output:
```
Installing types-cryptography (3.3.23): Failed

  DBusErrorResponse

  [org.freedesktop.DBus.Error.UnknownMethod] ('No such interface “org.freedesktop.DBus.Properties” on object at path /org/freedesktop/secrets/collection/login',)

  at /opt/anaconda3/lib/python3.9/site-packages/jeepney/wrappers.py:214 in unwrap_msg
      210│     This is to be used with replies to method_call messages, which may be
      211│     method_return or error.
      212│     """
      213│     if msg.header.message_type == MessageType.error:
    → 214│         raise DBusErrorResponse(msg)
      215│ 
      216│     return msg.body
      217│ 

The following error occurred when trying to handle this error:

  ItemNotFoundException

  Item does not exist!

  at /opt/anaconda3/lib/python3.9/site-packages/secretstorage/util.py:49 in send_and_get_reply
       45│              try:
       46│                      return self._connection.send_and_get_reply(msg, unwrap=True)
       47│              except DBusErrorResponse as resp:
       48│                      if resp.name in (DBUS_UNKNOWN_METHOD, DBUS_NO_SUCH_OBJECT):
    →  49│                              raise ItemNotFoundException('Item does not exist!') from resp
       50│                      elif resp.name in (DBUS_SERVICE_UNKNOWN, DBUS_EXEC_FAILED,
       51│                                         DBUS_NO_REPLY):
       52│                              data = resp.data
       53│                              if isinstance(data, tuple):

The following error occurred when trying to handle this error:

  PromptDismissedException

  Prompt dismissed.

  at /opt/anaconda3/lib/python3.9/site-packages/secretstorage/collection.py:145 in create_collection
      141│      if len(collection_path) > 1:
      142│              return Collection(connection, collection_path, session=session)
      143│      dismissed, result = exec_prompt(connection, prompt)
      144│      if dismissed:
    → 145│              raise PromptDismissedException('Prompt dismissed.')
      146│      signature, collection_path = result
       61│                 collection = secretstorage.get_default_collection(bus)
  • Installing pyasn1 (0.4.8): Failed

  DBusErrorResponse

  [org.freedesktop.DBus.Error.UnknownMethod] ('No such interface “org.freedesktop.DBus.Properties” on object at path /org/freedesktop/secrets/collection/login',)

[... 901 lines cut ...]
      143│      dismissed, result = exec_prompt(connection, prompt)
      144│      if dismissed:
    → 145│              raise PromptDismissedException('Prompt dismissed.')
      146│      signature, collection_path = result
      147│      assert signature == 'o'
      148│      return Collection(connection, collection_path, session=session)
      149│ 

The following error occurred when trying to handle this error:

  InitError

  Failed to create the collection: Prompt dismissed..

  at /opt/anaconda3/lib/python3.9/site-packages/keyring/backends/SecretService.py:63 in get_preferred_collection
       59│                 collection = secretstorage.Collection(bus, self.preferred_collection)
       60│             else:
       61│                 collection = secretstorage.get_default_collection(bus)
       62│         except exceptions.SecretStorageException as e:
    →  63│             raise InitError("Failed to create the collection: %s." % e)
       64│         if collection.is_locked():
       65│             collection.unlock()
       66│             if collection.is_locked():  # User dismissed the prompt
       67│                 raise KeyringLocked("Failed to unlock the collection!")
```

aorgazf, 2022-10-27:
I tried installing qibocal with `pip install -e .`but when running `qq-live` it throws another error as if pint hadn't been installed properly:
```
Traceback (most recent call last):
  File "/home/users/alvaro.orgaz/.conda/envs/qiboteam/bin/qq-live", line 5, in <module>
    from qibocal import live_plot
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/__init__.py", line 2, in <module>
    from .cli import command, live_plot, upload
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/__init__.py", line 3, in <module>
    from ._base import command, live_plot, upload
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/_base.py", line 14, in <module>
    from qibocal.cli.builders import ActionBuilder
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 9, in <module>
    from qibocal import calibrations
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/calibrations/__init__.py", line 2, in <module>
    from qibocal.calibrations.characterization.allXY import *
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/calibrations/characterization/allXY.py", line 6, in <module>
    from qibocal import plots
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/plots/__init__.py", line 2, in <module>
    from qibocal.plots.heatmaps import *
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/plots/heatmaps.py", line 7, in <module>
    from qibocal.data import DataUnits
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/data.py", line 9, in <module>
    import pint_pandas
  File "/home/users/alvaro.orgaz/.conda/envs/qiboteam/lib/python3.9/site-packages/pint_pandas/__init__.py", line 3, in <module>
    from .pint_array import PintArray, PintType
  File "/home/users/alvaro.orgaz/.conda/envs/qiboteam/lib/python3.9/site-packages/pint_pandas/pint_array.py", line 22, in <module>
    from pint.quantity import _Quantity
ModuleNotFoundError: No module named 'pint.quantity'

```

andrea-pasquale, 2022-10-27:
Hi @aorgazf,
regarding the first comment I think that this is the same problem the Boulos was having. It happened to me also once and to @DavidSarlle.
It seems that there is a problem between poetry and a program called `keyring` that usually run in linux devices.
The easiest solution to avoid this error is to disable it by typing:
```sh
keyring --disable
```
If you try to install the package again using poetry it should work.
It is not a problem related to `qibocal` itself so, apart from writing a warning in the documentation, we cannot do much else.
A possible solution at least for `sadyaat` would be to disable this program by default for all users.

Regarding the second comment I have no idea why this happens. I will look more into that as soon as I'm back.

PR #135 (2022-11-01): Update deps
Closes #132. Apparently there was an issue between the `pint` and `pint_pandas` [here](https://github.com/hgrecco/pint-pandas/issues/145), that has been fixed shortly after [here](https://github.com/hgrecco/pint-pandas/pull/133).

@aorgazf let me know if now everything works.

Moreover, now qibocal will install the latest main of qibo and not the latest release to avoid raising an error concerning the versions attribute in the backends. @scarrazza as soon as we have the first release of qibocal we should probably make a new release of qibo to avoid such issues.

The main issue was  [... cut]
closed 2022-11-01 by scarrazza