qiboteam/qibocal #136: Action builder for multiple qubits on quantum hardware UnboundLocalError

wilkensJ, 2022-10-31:
When running a job on the qpu5q the index.html page cannot be build because of that line in builders.py:
https://github.com/qiboteam/qibocal/blob/177b2a5613a9b25e2a47788b25b6f810986ac428/src/qibocal/cli/builders.py#L168-L169
The error looks like this
``` txt
File "/nfs/users/jadwiga.wilkens/qibocal/src/qibocal/cli/builders.py", line 169, in _execute_single_action
    self.update_platform_runcard(qubit, routine.__name__)
UnboundLocalError: local variable 'qubit' referenced before assignment
```
Commenting it out works.

andrea-pasquale, 2022-10-31:
Thanks for opening this issue @wilkensJ.
Indeed this is a bug. 

The general idea is that after we run a RB we can add the fidelity to the platform parameters, correct?
By looking at the runcard https://github.com/qiboteam/qibocal/blob/245519b56bc8832f790e144fdf2a8ec961b2fcef/runcards/actions_qq_5q.yml#L16 you are computing the fidelity of qubit 2, which corresponds to the parameter `active_qubit`.
A quick fix will be to pass the parameter `active_qubit` instead of `qubit` in `update_platform_runcard`.

However, in order for the `update_platform` method to work you need to save to file (usually we call the file `f"fit_q{qubit}"`) the parameters of the fit. Now they are computed on the fly when plotting.

An easy way to save the fitting parameters to file is to yield them directly from the calibration routines:
https://github.com/qiboteam/qibocal/blob/177b2a5613a9b25e2a47788b25b6f810986ac428/src/qibocal/calibrations/characterization/resonator_spectroscopy.py#L50

I'll suggest you to implement the fitting in way, so that I can quickly fix this issue.

scarrazza, 2023-01-23:
Shall we close this issue?

wilkensJ, 2023-01-24:
I did not try to run a protocol for a single qubit on the 5 qubit chip again, so I don't know if that error is still a problem.

andrea-pasquale, 2023-01-24:
At least in main this is fixed since after the implementation of the multiplex readout all the plotting and the fitting functions have been slightly changed. Therefore I suggest to close this issue.

closed 2023-01-24 by scarrazza