qiboteam/qibocal #141: Pandas error when adding an iterable to data.Data object

wilkensJ, 2022-11-10:
When adding an array or list to an Data object an error is raised.

``` python
from qibocal.data import Data
import numpy as np

mydata = Data('testpandas', quantities=[1, 2])
test_data = ['hello', 'Andrea']
mydata.add({
    1:test_data[0], 2:test_data[1]
})
```
Works fine, but if the data to add would be an array for example:
``` python
from qibocal.data import Data
import numpy as np

mydata = Data('testpandas', quantities=[1, 2])
test_data = [np.array([2,5]), np.array([10,20])]
mydata.add({
    1:test_data[0], 2:test_data[1]
})
```
an error is raised
``` txt
  File "/nfs/users/jadwiga.wilkens/qibocal/src/qibocal/data.py", line 273, in add
    self.df.loc[l, key] = value
  File "/home/users/jadwiga.wilkens/.conda/envs/testenv/lib/python3.9/site-packages/pandas/core/indexing.py", line 818, in __setitem__
    iloc._setitem_with_indexer(indexer, value, self.name)
  File "/home/users/jadwiga.wilkens/.conda/envs/testenv/lib/python3.9/site-packages/pandas/core/indexing.py", line 1795, in _setitem_with_indexer
    self._setitem_with_indexer_split_path(indexer, value, name)
  File "/home/users/jadwiga.wilkens/.conda/envs/testenv/lib/python3.9/site-packages/pandas/core/indexing.py", line 1850, in _setitem_with_indexer_split_path
    raise ValueError(
ValueError: Must have equal len keys and value when setting with an iterable
```

scarrazza, 2023-01-23:
Shall we close this issue?

wilkensJ, 2023-01-24:
Well the error still occurs when running the above code I stated. I build a workaround, namely initiating an own pandas data frame and storing it directly as attribute in an empty ``Data`` object.
I can imagine users wanting to store arrays/lists in this ``Data`` object (as I wanted it), but I don't know how to fix the error.

andrea-pasquale, 2023-01-24:
Now both `Data` and `DataUnits` have a method called [`load_data_from_dict`](https://github.com/qiboteam/qibocal/blob/e4c7eaaf4f1cf2f50cca0041d93155563b475454/src/qibocal/data.py#L121) that does exactly that. So I think that we can close this issue.

closed 2023-01-24 by scarrazza