qiboteam/qibocal #142: new installation issue on windows

aorgazf, 2022-11-10:
Hi Andrea,
I've tried setting up a new environment and I get the following error:

```
  • Installing pastel (0.2.1)
  • Installing sphinx (4.5.0)
  • Installing uncertainties (3.1.7)

  OSError

  [WinError 1314] A required privilege is not held by the client: '../../../../../examples/3_tangle/README.md' -> 'C:\\Users\\Alvaro\\anaconda3\\envs\\qibo11\\src\\qibo\\doc\\source\\code-examples\\tutorials\\3_tangle\\README.md'

  at ~\anaconda3\envs\qibo11\lib\site-packages\dulwich\index.py:606 in symlink
       602│                 filename)
       603│
       604│     def symlink(src, dst, target_is_directory=False, *, dir_fd=None):
       605│         try:
    →  606│             return os.symlink(
       607│                 src, dst, target_is_directory=target_is_directory,
       608│                 dir_fd=dir_fd)
       609│         except PermissionError as e:
       610│             raise WindowsSymlinkPermissionError(
```

the code executed is:

```bash
conda create -n qibo11 python=3.9
conda activate qibo11
git clone https://github.com/qiboteam/qibo
cd qibo
pip install -e .
cd ..
pip install poetry
pip install pre-commit
git clone https://github.com/qiboteam/qibocal
cd qibocal
poetry install
```

aorgazf, 2022-11-10:
If I install qibocal using pip it works:
```bash
conda create -n qibo11 python=3.9
conda activate qibo11
git clone https://github.com/qiboteam/qibo
cd qibo
pip install -e .
cd ..
git clone https://github.com/qiboteam/qibocal
cd qibocal
pip install . -e
cd ..
git clone https://github.com/qiboteam/qibolab
cd qibolab
pip install -e .[tiiq]
```

andrea-pasquale, 2022-11-10:
Thanks for opening this issue @aorgazf.
I'm able to reproduce this issue on windows.
I'll take a look.

aorgazf, 2022-11-11:
Installing with pip install the code later complains about `lmfit` library missing (not sure if this is caused for not using poetry)

andrea-pasquale, 2022-11-11:
@aorgazf may I ask you which branch of qibocal and qibolab are you using?
Recently we moved some dependencies from qibolab to qibocal `limfit` being one of those.

aorgazf, 2022-11-14:
Hi, I started with a fresh install of both repositories with `git clone`, so I was using `main` branch on both.

PR #172 (2022-12-06): Fixing windows installation
Closes #142.

This PR fixes the latest installation problems with windows.
Windows was complaining about some symbolic links which are in Qibo.
This issue seems to appear only if in  the `pyproject.toml` we install the latest main of Qibo, instead if we install the latest release of Qibo this issue is no longer present. I think that there is nothing wrong with the latest main of Qibo, probably poetry has some issues when installing "from source" projects that have symbolic links.
The fix proposed in this PR consists in changing the Qibo dependency from the latest ma [... cut]
closed 2022-12-06 by scarrazza