qiboteam/qibocal #146: Continuous measurements of T1 and T2 on two qubits

Boulos-Alfakes, 2022-11-11:
We are observing variations in T1 and T2 as a function of time. If possible to develop a function that runs extended scans to monitor those parameters for two different qubits simultaneously.

Thanks.

andrea-pasquale, 2022-11-11:
Hi @Boulos-Alfakes, thanks for opening this issue.

If I understand correctly for continuous scans you mean that after executing the function that calculates T1 we should re-execute the function multiple times, correct? Theoretically this could be done by using a large number of software averages, or in order to keep the job running we can have a while loop. Then, we can also specify in the data structure when each T1 value was computed.

Instead of using software averages we can even write a script that will launch qq continuously. This approach should be the simplest one. With this you will have a single qq report for each software average.

About running an experiment on two qubits simultaneously, we need to check that everything is working correctly but theoretically it should be feasible.

Boulos-Alfakes, 2022-11-14:
Dear Andrea,

Thank you for following up, and your prompt response. I am ok with whatever approach is easier for your to implement, as long as I can plot T1 as a function of time and day. With the current parameters am using, T1 scan takes around 4-5 minutes. That should give 12 - 15 points per hour. It would be easier to have them in 1 report, but if it is hard to implement, then don't worry about it, I can read the different reports and record the values manually.

Cheers,
Boulos

stavros11, 2022-11-14:
Regardless of whether we are using multiple or a single job, I think it makes more sense to have a single file with datetime, T1 and T2 entries, as it will be easier to read. That being said, even if you use different jobs for each T1 measurement, you can easily aggregate the results to a single file with some post-processing script, using the datetimes (for start and end) from the meta.yml.