qiboteam/qibocal #158: From factory to production

scarrazza, 2022-11-18:
We are missing documentation and scripts to simplify the process of making a new qubit operative for a known platform setup.
Let me put down some points which should be reflected in #103 and autocalibration.
- generate default runcard (e.g. iteractive script which asks how many qubits, the supposed frequencies) and dumps a runcard following qibolab syntax.
- execute `qq` with a default runcard which should generate an acceptable calibration runcard.
- document the process

@andrea-pasquale may I ask you to collect the available and missing features to achieve such workflow?

andrea-pasquale, 2022-11-21:
I agree @scarrazza.

> * generate default runcard (e.g. iteractive script which asks how many qubits, the supposed frequencies) and dumps a runcard following qibolab syntax.

For default runcard you mean the yaml that it will be passed to `qq` or the platform runcard in qibolab?
I don't think that we can easily code the whole platform runcard through an interactive script, especially the instruments part.
So I think that the instruments part should already be provided by the people in the lab when they set up a new platform.

> * execute `qq` with a default runcard which should generate an acceptable calibration runcard.

The default runcard should be platform agnostic? If not, we can use the previous script to generate a runcard where the different ranges are computed after specifying the supposed frequencies.

Let me know what you think.

stavros11, 2022-11-21:
> I don't think that we can easily code the whole platform runcard through an interactive script, especially the instruments part. So I think that the instruments part should already be provided by the people in the lab when they set up a new platform.

I think that as qibolab is now, the platform assumes a specific structure for the runcard. Therefore this "script" could be a platform method that generates a compatible runcard and the user can specify the setup specific parameters (instrument IPs, fabrication values for resonator and qubit frequencies, etc.). Another advantage of this is that it will require standardizing the runcard format of each platform.

That being said, I do not think such script is necessary. As Andrea said, the experimentalist can just create a runcard that has the proper format and the frequency values expected from fabrication and start from there. Creating such runcard is rather simple anyway, just copy the existing one and change some values. What could help is adding the option to pass a platform runcard to qq, because now it is using the default runcard from qibolab, so the user would have to dump the new runcard there.

Let me know what you think.

andrea-pasquale, 2022-11-21:
> What could help is adding the option to pass a platform runcard to qq, because now it is using the default runcard from qibolab, so the user would have to dump the new runcard there.

At the beginning we had this option already. It should be straightforward to add this feature.

scarrazza, 2022-11-21:
Indeed, I was considering "known platform setup" so the user can clone a copy of existing runcard and the script asks for basic replacements. Anyways, the script is not mandatory, but we need documentation about these steps.

closed 2023-01-11 by scarrazza