qiboteam/qibocal #167: Jadwigas small wishlist for Qibocal

wilkensJ, 2022-11-29:
- [ ] When only executing simulations it should not be necessary to load qibolab.
- [ ] The import of all the modules in ``__init__.py`` in the folder src/qibocal/calibrations/characterization does not allow for calling a function in different modules the same. Also, if there is an error in a module which is not used by the user it breaks everything.
- [ ] Change the folder structure (@ingoroth)
- [ ] Think of a nice visualization of RB reports, I suggest a dashboard.

andrea-pasquale, 2022-11-30:
Thanks @wilkensJ.

> [ ] When only executing simulations it should not be necessary to load qibolab.

I agree. This should be pretty easy to implement. Right now, qibolab is a dependency of qibocal (when you install qibocal you will install the main branch qibolab). We could remove it and raise an error if the user tries to use a routine that requires qibolab and qibolab is not installed. You should discuss this during the qibo meeting or in a separate meeting.

> [ ] The import of all the modules in `__init__.py` in the folder src/qibocal/calibrations/characterization does not allow for calling a function in different modules the same. Also, if there is an error in a module which is not used by the user it breaks everything.

This is related to the point above, I can take care of this.

> [ ] Change the folder structure (@ingoroth)

I am fine with changing the folder structure, but he should all agree on a specific folder structure and avoid to change it multiple times.

wilkensJ, 2022-12-19:
# What the action builder should be able to do:
@andrea-pasquale here a little sketch for what the action builder should do in the end.
Use the `standadrb` model as an example what the action builder should then do on an abstract level.
## Input:
Module name and the parameters like qubits, depths, ... (plus maybe the error model used in a simulation, not sure how this will look like)
```yml
protocols:
  standardrb:
    nqubit: 1
    depths: [0,1, 3,5,7,10,15,20]
    runs: 5
    nshots: 1024
    noise_params: [0.01, 0.01, 0.01]
```
## Action:
Allocates the classes and functions associated to the module.
**Module specific**:
```python
Factory =  standardrb.SingleCliffordsInvFactory
Experiment = standardrb.StandardRBExperiment
noise = ... # I don't know yet how to integrate noise models nicely
validation_function =  standardrb.validate_simulation
```
**Generic**:
```python
# Initiate the circuit factory and the Experiment object.
factory = Factory(nqubit, depths, runs, qubits=qubit)
experiment = Experiment(factory, nshots, noisemodel=noise)
# Execute the experiment.
experiment.execute()
data = Data()
data.df = experiment.dataframe
yield data
data_validation = Data("validate_simulation", quantities=["validation"])
data_validation.add({"validation": validation_function(experiment)})
yield data_validation
```
## Plot
**Module specific**:
```python
Experiment = standardrb.StandardRBExperiment
analyze = standardrb.analyze 
```
**Generic**:
```python
experimentpath = f"{folder}/data/{routine}/"
experiment = Experiment.load(experimentpath)
fig = analyze(experiment)
data = Data.load_data(folder, routine, "pickle", "validation_simulation")
validation = data.df.to_numpy()[0, 0]
fig.add_annotation(
    dict(
        font=dict(color="black", size=12),
        x=0,
        y=-0.20,
        showarrow=False,
        text="Validation parameters: {:.3f}".format(validation),
        textangle=0,
        xanchor="left",
        xref="paper",
        yref="paper",
    )
)
return fig
```

wilkensJ, 2022-12-20:
Update from Zoom meeting:
```python
class Protocol:

    def __init__(self, module, noise = None) -> None:
        self.module = module
        self.noise = noise

    def execute(
            self,
            qubit: list,
            depths: list,
            runs: int,
            nshots: int,
            nqubit: int = None) -> None:
        # Initiate the circuit factory and the Experiment object.
        factory = self.module.Factory(nqubit, depths, runs, qubits=qubit)
        experiment = self.module.Experiment(factory, nshots, noisemodel=self.noise)
        # Execute the experiment.
        experiment.execute()
        data = Data()
        data.df = experiment.dataframe
        yield data
        data_validation = Data("validate_simulation", quantities=["validation"])
        data_validation.add({"validation": self.module.validation_function(experiment)})
        yield data_validation
```

aorgazf, 2023-05-31:
Hi @wilkensJ,
I was trying to clean up open issues, could you please update the status of this one?
Perhaps close it and reopen separate issues for those items not addressed.

closed 2023-05-31 by aorgazf

wilkensJ, 2023-05-31:
Hello @aorgazf,
thank you for keeping that in mind, I think these points are outdated since the auto calibration was introduced, it can stay closed :)