qiboteam/qibocal #198: Failed to install with poetry on windows

maxhant, 2023-01-13:
Using python 3.9, here is the error I get on a fresh conda env:

```
(qm_qibo) C:\Users\maxime.hantute\OneDrive - Technology Innovation Institute\Repositories\maximelab\qm_qibo>python -m pip install poetry
Collecting poetry
  Downloading poetry-1.3.2-py3-none-any.whl (218 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 218.9/218.9 kB 6.7 MB/s eta 0:00:00
Collecting keyring<24.0.0,>=23.9.0
  Downloading keyring-23.13.1-py3-none-any.whl (37 kB)
Collecting requests-toolbelt<0.11.0,>=0.9.1
  Downloading requests_toolbelt-0.10.1-py2.py3-none-any.whl (54 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 54.5/54.5 kB ? eta 0:00:00
Collecting lockfile<0.13.0,>=0.12.2
  Using cached lockfile-0.12.2-py2.py3-none-any.whl (13 kB)
Collecting poetry-plugin-export<2.0.0,>=1.2.0
  Using cached poetry_plugin_export-1.2.0-py3-none-any.whl (10 kB)
Collecting virtualenv!=20.4.5,!=20.4.6,<20.16.6,<21.0.0,>=20.4.3
  Using cached virtualenv-20.16.5-py3-none-any.whl (8.8 MB)
Collecting poetry-core==1.4.0
  Using cached poetry_core-1.4.0-py3-none-any.whl (546 kB)
Collecting filelock<4.0.0,>=3.8.0
  Downloading filelock-3.9.0-py3-none-any.whl (9.7 kB)
Collecting pexpect<5.0.0,>=4.7.0
  Using cached pexpect-4.8.0-py2.py3-none-any.whl (59 kB)
Collecting importlib-metadata<5.0,>=4.4
  Using cached importlib_metadata-4.13.0-py3-none-any.whl (23 kB)
Collecting dulwich<0.21.0,>=0.20.46
  Downloading dulwich-0.20.50-cp39-cp39-win_amd64.whl (472 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 472.4/472.4 kB 7.5 MB/s eta 0:00:00
Collecting html5lib<2.0,>=1.0
  Using cached html5lib-1.1-py2.py3-none-any.whl (112 kB)
Collecting tomlkit!=0.11.2,!=0.11.3,<1.0.0,>=0.11.1
  Using cached tomlkit-0.11.6-py3-none-any.whl (35 kB)
Collecting jsonschema<5.0.0,>=4.10.0
  Using cached jsonschema-4.17.3-py3-none-any.whl (90 kB)
Collecting crashtest<0.5.0,>=0.4.1
  Downloading crashtest-0.4.1-py3-none-any.whl (7.6 kB)
Collecting trove-classifiers>=2022.5.19
  Downloading trove_classifiers-2023.1.12-py3-none-any.whl (13 kB)
Collecting cleo<3.0.0,>=2.0.0
  Downloading cleo-2.0.1-py3-none-any.whl (77 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 77.3/77.3 kB 4.2 MB/s eta 0:00:00
Collecting urllib3<2.0.0,>=1.26.0
  Downloading urllib3-1.26.14-py2.py3-none-any.whl (140 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 140.6/140.6 kB 8.7 MB/s eta 0:00:00
Collecting tomli<3.0.0,>=2.0.1
  Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting pkginfo<2.0,>=1.5
  Downloading pkginfo-1.9.6-py3-none-any.whl (30 kB)
Collecting requests<3.0,>=2.18
  Downloading requests-2.28.2-py3-none-any.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.8/62.8 kB 3.3 MB/s eta 0:00:00
Collecting cachecontrol[filecache]<0.13.0,>=0.12.9
  Using cached CacheControl-0.12.11-py2.py3-none-any.whl (21 kB)
Collecting shellingham<2.0,>=1.5
  Downloading shellingham-1.5.0.post1-py2.py3-none-any.whl (9.4 kB)
Collecting platformdirs<3.0.0,>=2.5.2
  Downloading platformdirs-2.6.2-py3-none-any.whl (14 kB)
Collecting packaging>=20.4
  Downloading packaging-23.0-py3-none-any.whl (42 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.7/42.7 kB 2.2 MB/s eta 0:00:00
Collecting msgpack>=0.5.2
  Downloading msgpack-1.0.4-cp39-cp39-win_amd64.whl (62 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 62.5/62.5 kB ? eta 0:00:00
Collecting rapidfuzz<3.0.0,>=2.2.0
  Downloading rapidfuzz-2.13.7-cp39-cp39-win_amd64.whl (1.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 9.3 MB/s eta 0:00:00
Collecting six>=1.9
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting webencodings
[... 61 lines cut ...]
  EnvCommandError

  Command C:\Users\maxime.hantute\Anaconda3\envs\qm_qibo\python.exe -m pip install --disable-pip-version-check --isolated --no-input --prefix C:\Users\maxime.hantute\Anaconda3\envs\qm_qibo --upgrade --no-deps C:\Users\maxime.hantute\AppData\Local\pypoetry\Cache\artifacts\3e\99\e2\360c4ab5408a60c2f59e1f90b0e788ec410f1ebc0921724075288f9af9\charset_normalizer-2.1.1-py3-none-any.whl errored with the following return code 1, and output:
  Processing c:\users\maxime.hantute\appdata\local\pypoetry\cache\artifacts\3e\99\e2\360c4ab5408a60c2f59e1f90b0e788ec410f1ebc0921724075288f9af9\charset_normalizer-2.1.1-py3-none-any.whl
  Installing collected packages: charset-normalizer
    Attempting uninstall: charset-normalizer
      Found existing installation: charset-normalizer 3.0.1
      Uninstalling charset-normalizer-3.0.1:
        Successfully uninstalled charset-normalizer-3.0.1
  ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: 'C:\\Users\\maxime.hantute\\Anaconda3\\envs\\qm_qibo\\Lib\\site-packages\\~harset_normalizer\\md.cp39-win_amd64.pyd'
  Consider using the `--user` option or check the permissions.

2.1.1-py3-none-any.whl

  at ~\Anaconda3\envs\qm_qibo\lib\site-packages\poetry\utils\pip.py:58 in pip_install
       54│
       55│     try:
       56│         return environment.run_pip(*args)
       57│     except EnvCommandError as e:
    →  58│         raise PoetryException(f"Failed to install {path.as_posix()}") from e
       59│

(qm_qibo) C:\Users\maxime.hantute\OneDrive - Technology Innovation Institute\Repositories\maximelab\qm_qibo\qibocal>git branch
* main
```

andrea-pasquale, 2023-01-13:
It seems like an issue with the installation of `poetry` not with the installation of  `qibocal`.
Does it work if you install qibocal with pip using `pip install -e .`?

maxhant, 2023-01-13:
Yes it works, that's what I did later

closed 2023-01-14 by andrea-pasquale