qiboteam/qibocal #229: Intermediate saving of HTML reports

aorgazf, 2023-02-01:
Hi,
Currently, the html reports are only generated and saved when the qibocal routine finishes. Sometimes the routine does not complete or is stopped by the user. Ideally intermediate html reports are saved.

stavros11, 2023-02-16:
@andrea-pasquale @scarrazza this would be very useful to have, particularly after merging the sweepers, as the report can serve as a better alternative to qq-live and it is much faster to load too. Now we dump the report only once in the end, ideally we should dump it after every routine.

This is linked to #232, however as pointed out in https://github.com/qiboteam/qibocal/pull/232#issuecomment-1430884114 I do not think it is addressed there yet.

Also, there are many cases where the data acquisition works, but the report generation fails, for example due to a bug in a plot or a fitting fail, so we end up with a folder that has all data but no report. Do you think it would be useful to have a `qq-*` command to regenerate a report using existing data? This would also be useful when one wants to change something in a plot only, but not necessarily rerun the whole data acquisition.

scarrazza, 2023-02-16:
> Do you think it would be useful to have a qq-* command to regenerate a report using existing data?

Yes, I think is a good idea to wrap the report generator function in another program. In fact, if there is agreement about a future deprecation of qq-live, we could introduce operational modes for qq, e.g. data-acquisition, post-processing, following a similar pattern already used by RB.

andrea-pasquale, 2023-02-16:
I also agree. We definitively need to dump the report after every routine.

> In fact, if there is agreement about a future deprecation of qq-live, we could introduce operational modes for qq, e.g. data-acquisition, post-processing, following a similar pattern already used by RB.

Yes, once we merge the #239 we can think about changing all the routines in an `Experiment` object (or something similar) to be able to define different sub parts of each routine as @scarrazza.

For now I can work on a PR just to implement #232.

PR #245 (2023-02-17): Dump report after each routine
Closes #229.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [ ] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
closed 2023-02-17 by scarrazza