qiboteam/qibocal #253: Ignore Sphinx generat-ed/-able autodoc

AleCandido, 2023-02-20:
The files in https://github.com/qiboteam/qibocal/tree/b8fa388befd51245a604fe721e19f9d3a6ab9cf4/doc/source/api-reference contain a lot of calls to `autodoc`, but in principle those can be completely generated by running [`sphinx-apidoc`](https://www.sphinx-doc.org/en/master/man/sphinx-apidoc.html) on the source code.

So, I'd rather propose to move the human edited part of those files inside the source code itself (or other docs sections in case it is too complex for source code, we definitely do not want to pollute too much the source), and remove these files from the repo.
In particular, to simplify the process of running `sphinx-apidoc`, we can completely avoid the call to the CLI, and make it part of the regular Sphinx build just hooking that inside the `doc/source/conf.py` file, like described in the following comment https://github.com/readthedocs/readthedocs.org/issues/1139#issuecomment-312626491.

andrea-pasquale, 2023-02-21:
Thanks for pointing this out @AleCandido.
I understand that in order to maintain the documentation this option is better. 
Personally I like the approach where we jump from human edited parts to `sphinx-apidoc` parts (maybe I'm bias because when I started writing the documentation I was mainly looking at the qibo documentation).
I imagine that another option would be to move the human edited parts somewhere else and include only reference to the api reference section, right?
Having said that I'm still a bit unsure on how to proceed.
Maybe it is also worth to look at other projects documentation to take inspiration.
Do you know if there is a more or less standard approach for writing the documentation?

AleCandido, 2023-02-21:
> Do you know if there is a more or less standard approach for writing the documentation?

I believe there are many possible options and corners to deal with according to the structure of the program. E.g. using Sphinx for packages that are not pure Python is not always straightforward. But this is not `qibocal` case :)

Another approach we may decide to follow (and might even be worth considering in view of the overall website) would be to limit the Sphinx usage to the very minimal subset of docs made by docstrings, and keep all the rest of the documentation in Markdown files and Jupyter notebooks outside Sphinx, such that we can render them as more integrated pages on the website (built directly by Next.js).
Just to give you a practical example: instead of building and deploying all the pages in this project itself, we could trigger a workflow on the website repository, that pulls the content of a dedicated folder with markdown and notebooks, plus a pre-built minimal Sphinx pages, and then regenerate the whole Qibo website (with suitable caching for unchanged parts, of course).

> I imagine that another option would be to move the human edited parts somewhere else and include only reference to the api reference section, right?

However, any more involved strategy is beyond the scope of the issue: my **goal** here was just to avoid listing in the PR "Files changed" tab all the `.. automodule` directives, since they can be automatically generated, and thus they don't have to be committed to the repo. And to do this, you just need to move the "human content" out of `api-reference`, and keep this fully generated.
Already this is low priority, it is just to clean up PRs. Any plan beyond this is definitely going in a separate PR and concerns the website development itself.

In the end, we do not always "standard practices", because there are tasks that might be done in several different ways, and different people/projects make different choices. We just need a sensible and motivated approach :)

andrea-pasquale, 2023-09-17:
I believe that this can be closed since we merged #511.

closed 2023-09-17 by andrea-pasquale