qiboteam/qibocal #254: Dropping pint(-pandas)

andrea-pasquale, 2023-02-22:
As already discussed in #249 we are planning to drop the `pint-pandas` dependency.
The main motivation is that, although it is interesting to have a mechanism to store units of measure and change them directly on the fly, this comes at the cost of having a lot of complications inside the code especially regarding data saving/loading.
By not using `pint` we will be able to simplify several parts of the code including the calibration routines.
As far as I understood we are planning to use custom `dtypes` following the same approach that we used in qibolab for the [`ExecutionResults`](https://github.com/qiboteam/qibolab/blob/1ae75a8a21aef417f589cf95325edcac0b9d135a/src/qibolab/result.py#L41) object.

We can use this issue to further elaborate on which is the best solution to adopt for qibocal.

aorgazf, 2023-02-23:
Hi,
I would like to add another reason for dropping `pint-pandas`, which is that it dramatically slows the performance of pandas.
The problems we have with the plotting functions would go away just by `de-quantifying` the dataframe before working with it:
```python
data.df[quantities] = data.df[data.quantities].pint.dequantify()
```

andrea-pasquale, 2023-06-01:
I've a proposal for a new data structure when storing the data acquired from a calibration routines.
Looking at the calibration routines I think that we need a structure that should be able to store

1. Qubit(Pair) specific quantities which may include voltages, frequencies, amplitudes...
2. Non Qubit(Pair) specific quantities which may include initial parameters (to be able to pass them to the results) and platform attributes which will be again processed in the results such as the resonator type.

The current approach involving pint is the following:

```py
class ResonatorSpectroscopyData(DataUnits):
    """ResonatorSpectroscopy acquisition outputs."""

    def __init__(self, resonator_type, power_level=None, amplitude=None):
        super().__init__(
            "data",
            {"frequency": "Hz"},
            options=["qubit"],
        )
        self._resonator_type = resonator_type
        self._power_level = power_level
        self._amplitude = amplitude

    @property
    def resonator_type(self):
        """Type of resonator"""
        return self._resonator_type

    @property
    def power_level(self):
        """Resonator spectroscopy power level"""
        return self._power_level

    @property
    def amplitude(self):
        """Readout pulse amplitude common for all qubits"""
        return self._amplitude
```
 in which 1 are stored through a `DataUnits` object which is basically a pandas dataframe + pint and 2 are currently not stored but they are used in runtime by the results object.

I know that the priority is to remove pint and keep things as simple as possible using for example `np.arrays`.
A possible solution could be the following:

```py
ResSpecType = np.dtype(
    [("freq", np.float64), ("msr", np.float64), ("phase", np.float64)]
)
"""Custom dtype for resonator spectroscopy."""

@dataclass
class ResonatorSpectroscopyData:
    """Data structure for resonator spectroscopy."""

    power_level: PowerLevel
    """Power regime of the resonator."""
    resonator_type: str
    """Resonator type."""
    amplitudes: list
    """Amplitudes provided by the user."""
    data: Dict[QubitId, npt.NDArray[ResSpecType]] = field(default_factory=dict)
    """Raw data acquired."""

    def load_qubit(self, qubit, freq, msr, phase):
        """Store output for single qubit."""
        ar = np.empty(freq.shape, dtype=ResSpecType)
        ar["freq"] = freq
        ar["msr"] = msr
        ar["phase"] = phase
        self.data[qubit] = np.rec.array(ar)

    @property
    def qubits(self):
        """Access qubits from data structure."""
        return [q for q in self.data]

    def __getitem__(self, qubit):
        return self.data[qubit]

    def save(self, path):
        """Store results."""
        self.power_level = self.power_level.name
        my_dict = asdict(self)
        my_dict.pop("data")
        with open(path / "conf.yml", "w") as outfile:
            yaml.dump(my_dict, outfile)

        for q in self.data:
            np.save(path / f"data_q{q}", self[q])
```
Where I do the following:

1. Introduce custom dtype for each routine (here for resonator spectroscopy I just need to store freq, volt and phase) and store data for each qubit using a rec array
2. Multiple qubits are collect through a dictionary `Dict[QubitId, npt.NDArray[ResSpecType]]` and they are added sequentially using `load_qubit` which follows more or less what we are doing now during the acquisition
3. When storing there is a yml ( we can choose whatever format we like) where I store data of the type 2 (global) and the data related to each qubit are stored currently using .npy files. I generate one file for each qubit.

This is just a prototype for now and I have a working version just for resonator specotroscopy in [this ](https://github.com/qiboteam/qibocal/tree/remove_pint) branch.

If you agree with this prototype I will propagate the changes for each routine.
Let me know what you think @scarrazza @stavros11 @AleCandido @Edoardo-Pedicillo @maxhant @aorgazf @DavidSarlle @rodolfocarobene @Jacfomg

P.S. Regarding units of measure currently I hardcode a specific one for each quantity. Each default unit of measure will be documented.

AleCandido, 2023-06-01:
> P.S. Regarding units of measure currently I hardcode a specific one for each quantity. Each default unit of measure will be documented.

I would keep this approach forever.

Units mainly distinguish different quantities, and secondarily they give you a scale factor with SI prefixes.
Here we do not need units to distinguish quantities: we have types and classes for them.

So, let's define some standard of units, and always use them. If you ever want to change them for plotting, this should be done at the most external level (or not at all in Qibocal).

AleCandido, 2023-06-01:
> This is just a prototype for now and I have a working version just for resonator specotroscopy in [this ](https://github.com/qiboteam/qibocal/tree/remove_pint?rgh-link-date=2023-06-01T07%3A27%3A33Z) branch.

Please, make a draft PR :)

andrea-pasquale, 2023-06-01:
> Please, make a draft PR :)

#372

Jacfomg, 2023-06-01:
Tested it and looks nice. I have some doubts on how this will behave for saving the data for something like the 2q gates were I have 3 qubits involved on the routine but I may only measure 1 or 2 at most.

andrea-pasquale, 2023-06-01:
Thanks for testing @Jacfomg!

> Tested it and looks nice. I have some doubts on how this will behave for saving the data for something like the 2q gates were I have 3 qubits involved on the routine but I may only measure 1 or 2 at most.

Keep in mind that you are free to customize the data structure for each routine.
For two qubit gates routines we could replace the data attribute with something like this:

```py
QubitPairId = Tuple[QubitId, QubitId]

...

data: data: Dict[QubitPairId, npt.NDArray[TwoQubitGatesRoutineType]] = field(default_factory=dict)
```

where you are also free to define `TwoQubitGatesRoutineType` as you wish.
I don't exactly know how we handle couplers in qibolab but I'm pretty sure that we can figure out a solution.

AleCandido, 2023-06-01:
Something like `QubitPairId` should also be part of Qibolab.

andrea-pasquale, 2023-06-22:
Closed by #372

closed 2023-06-22 by andrea-pasquale