qiboteam/qibocal #264: Automated runner: exceptional flow

AleCandido, 2023-03-06:
- [ ] agents (qubits) for multiplexing
- [ ] exceptional flow (and related test)
    - statuses (`Normal`, `Exceptional(id)`, `Broken`)
    - handlers.py file
- [ ] parameters passing (through function IDs)
- [ ] hook functions

andrea-pasquale, 2024-02-01:
@alecandido shall we close this given that #668 got merged?
Perhaps we can open an issue with what we should do next regarding this topic.

alecandido, 2024-02-01:
>  * [ ]  agents (qubits) for multiplexing

The missing part is multiplexing, but I feel that we're not yet ready for that. Maybe you could open an issue to keep track, but nothing else.

closed 2024-02-02 by andrea-pasquale

alecandido, 2024-02-02:
> Let me close the issue now and as soon we have a proposal I will open a dedicated issue.

It's fine

However, in general, you could open issues even to remind things that have to be done. If we had external users, I'd also have opened it in this case, to tell that is on-track (but we don't have any, and we know we'll come back to it).

andrea-pasquale, 2024-02-02:
Done https://github.com/qiboteam/qibocal/issues/702