qiboteam/qibocal #265: Automated runner: hook routines

AleCandido, 2023-03-06:
- [ ] hook functions

AleCandido, 2023-03-09:
After #237 will be merged, this will be the next up.

In principle, adding functions is an incremental task, but once agreed on the interface should be pretty smooth. I'd start from a single/couple of functions to test the rest.
The runcard update we'll have for sure to improve, and passing parameters mechanism we have to implement, but I already know how (dumping data we can keep at the end: once we have everything in memory, it's just a matter of picking a folder and insert a dump call).

I expect that for passing parameters I'll have to do something myself, but for the rest you can mostly take care of it @andrea-pasquale.
Of course I will help you, and we can discuss how to proceed before starting, however let me know if you disagree with part or the full plan :)

andrea-pasquale, 2023-03-10:
Thanks @AleCandido.

> In principle, adding functions is an incremental task, but once agreed on the interface should be pretty smooth. I'd start from a single/couple of functions to test the rest.

I completely agree. I'm having a look now.
I've noticed that the [`Routine` ](https://github.com/qiboteam/qibocal/blob/1d125d0b385cbb2d98e7e2c6718d34aa2ac0b70d/src/qibocal/auto/operation.py#L86) object , which should be where we will hook the routines I suppose, has two attributes: `acquisition` and `fit`.
Currently the routines that we have include also the fitting inside. 
I think that the easiest way to make things compatible is to split the fitting from the data acquisition in the routines.
This will enable us to have separate operational mode, as @scarrazza suggested here #229, including `data-acquisition` and `post-processing`.

As a first step I could the following:

- [ ] Detach fitting from calibration routines
- [ ] Refactor the `ActionBuilder` to be able to parse calibration as `Routine` objects
- [ ] Introduce commands for data-acquisition and post-processing

Regarding the second point we could also think about adopting already the runcard layout that you proposed in #237 with some adjustments. 

Let me know if you agree @scarrazza @stavros11 @AleCandido @Edoardo-Pedicillo.

AleCandido, 2023-03-10:
> I've noticed that the [`Routine` ](https://github.com/qiboteam/qibocal/blob/1d125d0b385cbb2d98e7e2c6718d34aa2ac0b70d/src/qibocal/auto/operation.py#L86) object , which should be where we will hook [...]

I made this on purpose: if you want to start by splitting, you're served, otherwise you can pass the fitting results as `Data`, and make `fit` an identity, that takes `Data` and just recasts them as `Results`, without doing any actual fit.

> Regarding the second point we could also think about adopting already the runcard layout that you proposed in #237 with some adjustments.

About this, I believe you could parse a runcard made in that fashion with very little extra effort (you take the `operation` value instead of the key, and the value of `parameters` instead of the current value).
In practice, you could also load with the current structure from `auto.runcard`, and take the information from there (but an `id` will be required for each step, that's the only difference, everything else is optional).