qiboteam/qibocal #275: Plotting unit of measure

rodolfocarobene, 2023-03-14:
In all the routines that I have used so far (resonator spectroscopy, qubit spectroscopy and rabi) the resulting plots have on the y axis an amplitude (MSR) expressed in  uV.
However, with QICK and the FPGA the result of a measurement is expressed in arbitrary ADC units so it's not trivial to convert them in V and I'm not sure it's even useful to try to find the right conversion factor. 
In my opinion, if these units are not important for the characterization (for the few routines I'm using they aren't), it would be nice if I could change the plotting units/conversion factor in the routine or to just use arbitrary units by default.

andrea-pasquale, 2023-03-16:
Thanks @rodolfocarobene for opening this issue.
Given that we are planning to drop soon pint (See #254), I think that using arbitrary should be the optimal solution.

andrea-pasquale, 2023-11-07:
Closing this as already present in #592

closed 2023-11-07 by andrea-pasquale