qiboteam/qibocal #289: Routine with sweepers not working for `Dummy`

andrea-pasquale, 2023-03-23:
After merging #195 when trying to execute a `resonator_spectroscopy` with the `DummyPlatform` (using qiboteam/qibolab#335). The code fails with the following error:

```bash
TypeError: to_dict() got an unexpected keyword argument 'average'
```
A quick fix would be to remove the average argument from `to_dict` inside the routines, but I would like to make sure that I don't break anything for the actual platforms. I will open the PR with a `do not merge` tag for now.

stavros11, 2023-03-23:
> A quick fix would be to remove the average argument from `to_dict` inside the routines, but I would like to make sure that I don't break anything for the actual platforms. I will open the PR with a `do not merge` tag for now.

Personally I don't like this solution, because then every time someone changes the default value of `average` in `to_dict` all routines in qibocal break. That's basically why I explicitly specify `average` in every routine, to make sure that the behavior is independent of the default value in qibolab (ideally I would remove the default completely to force the user to specify if they want average or no).

I am not sure why `to_dict` is different for the `DummyPlatform` though. Isn't this part of the `ExecutionResults` which are the same for all platforms?

andrea-pasquale, 2023-03-23:
I think that I understand the problem with `Dummy`.
In the sweep method of the `Dummy` there is this line

https://github.com/qiboteam/qibolab/blob/a0f6938da0fb7f91d7c1190923b147ee1c847d60/src/qibolab/instruments/dummy.py#L127

which basically changes the output type from [`ExecutionResults` ](https://github.com/qiboteam/qibolab/blob/a0f6938da0fb7f91d7c1190923b147ee1c847d60/src/qibolab/result.py#L41)to [`AverageExecutionResults` ](https://github.com/qiboteam/qibolab/blob/a0f6938da0fb7f91d7c1190923b147ee1c847d60/src/qibolab/result.py#L12), which does not accept the parameter `average`. This i

I remember that @AleCandido suggested to have a separate object for dealing with the case `average=True`.
I think that the mean reason for this was to pass the `average` parameter only once, in this case to the `sweep` method.

We have two possible solutions:

1. We drop `AverageExecutionResults` and we define explicitly in the `to_dict` method `average`.
2. We should update all the drivers in order to change the output type of the sweep method to `AverageExecutionResults` if `average=True`

Let me know what you think @stavros11

AleCandido, 2023-03-23:
> That's basically why I explicitly specify `average` in every routine, to make sure that the behavior is independent of the default value in qibolab (ideally I would remove the default completely to force the user to specify if they want average or no).

About this: why can't we just return a non-averaged result, and then if the users want the average, they will just call `.compute_average()` on the result?

@andrea-pasquale, btw, since `.compute_average()` takes only self, it can also be a `@property`, and we can call it `.average` :)

andrea-pasquale, 2023-03-23:
Thanks for the explanation @AleCandido.
> 
```py
avg = my_routine(...).compute_average()
```
Personally I like a lot this approach. It is for sure cleaner than doing this directly inside the `sweep` method.
@stavros11 if you agree we can opt for this solution.

rodolfocarobene, 2023-03-24:
For qick platforms I don't think that there is much difference.
I'm also in favour of just having the non averaged results returned

Jacfomg, 2023-03-24:
Yes @andrea-pasquale, I don't know the exact limit but I expect that averaging on the device compared to single shot will allow the device to rrun longer sweeps before running out of memory or just being faster than returning all data instead of a single value can be an overhead if we plan to do millions of shots as some people from theory expect. Also the averaging will be faster on the device than having to do it ourselves on qibocal I guess.

stavros11, 2023-03-24:
> Actually I think that it could have some value to return also an averaged result, given that some instruments are able to return directly the averaged values. @stavros11 @rodolfocarobene @Jacfomg this is the case for your platforms, right?

Indeed, at least for QM when `platform.sweep(average=True)` is used (which is the default for `platform.sweep`) the average over shots is performed in the instrument and only averaged values are returned. I guess a motivation for this is if you are doing a very fine 2D sweep and you are only interested in averages, it makes more sense to do the average on the device, otherwise it will have to return a 3D tensor which may be big and slow down the communication (and may also cause memory issues, although we have not really benchmark this).

stavros11, 2023-03-24:
> Moreover, I'd rather drop the `average` argument from `to_dict()`. If you want an averaged dict, the way to go should be `execres.compute_average().to_dict()` (and if possible, it would be even better to avoid using `dict`s as much as possible, and just work with the object, or the underlying array/s).

I agree with replacing `.to_dict(average=True)` wtih something like `execres.compute_average().to_dict()` or even `execres.average.to_dict()` if we make `average` a property of `ExecutionResults`.
Regarding dicts, I believe we currently use them mainly to populate the DataFrame used in qibocal routines. I am not sure if you (@andrea-pasquale @AleCandido) concluded in a better solution for data storage in qibocal, but when we were discussing with Andrea last week it was not very trivial how to replace the DataFrame.

Now regarding the current issue, I believe an acceptable solution would be:
* `platform.execute_pulse_sequence` returns `ExecutionResults`,
* `platform.sweep(average=False)` returns `ExecutionResults`,
* `platform.sweep(average=True)` returns `AveragedResults`.

At least my idea of `average=True` instruction in `sweep` is to do the average in the instrument, which means you don't even have access to non-averaged data. If Qick cannot perform this average we may need to rethink how to make it compatible with all platforms.

andrea-pasquale, 2023-03-24:
Thank you @stavros11. I was thinking the same.

> I agree with replacing `.to_dict(average=True)` wtih something like `execres.compute_average().to_dict()` or even `execres.average.to_dict()` if we make `average` a property of `ExecutionResults`. [...]

Not yet, from my side I want add first all the routine to tree with current data structures.

I will open a PR to fix the behavior of the sweep in qibolab. Thanks everyone for the quick feedback!

rodolfocarobene, 2023-03-24:
Also qick can perform averages and it's the default behaviour. 
But since it is done on hardware in any case, since the current implementation of the driver is on-board , there is not much difference is resources needed and I don't think that slow communication could be a problem.

> 
>     platform.execute_pulse_sequence returns ExecutionResults,
>     platform.sweep(average=False) returns ExecutionResults,
>     platform.sweep(average=True) returns AveragedResults.
> 

In any case the driver is now doing exactly this so any change would be immediately compatible with all platforms without any problem

stavros11, 2023-03-24:
Thanks for the responses.

> I will open a PR to fix the behavior of the sweep in qibolab. Thanks everyone for the quick feedback!

Note that the QM driver also needs to be updated as it is currently returning `ExecutionResults` in all cases (even when doing the average on the device). I can also fix it later today but feel free to do it if you want. I am not sure about qblox.

andrea-pasquale, 2023-03-24:
> Note that the QM driver also needs to be updated as it is currently returning `ExecutionResults` in all cases (even when doing the average on the device). I can also fix it later today but feel free to do it if you want. I am not sure about qblox.

Yeah, I expected that probably also the other drivers will need to updated. I can try to change it myself and I will ask if I need any help. 

Regarding qblox know we only have a software sweep so theoretically the same fix should work both for `Dummy` and `Qblox`.

AleCandido, 2023-03-24:
> Now regarding the current issue, I believe an acceptable solution would be:
>   * `platform.execute_pulse_sequence` returns `ExecutionResults`,
>   * `platform.sweep(average=False)` returns `ExecutionResults`,
>   * `platform.sweep(average=True)` returns `AveragedResults`.

For me this is a good enough solution, I was about to propose the same.
The rationale should be that if the `Union` is forced by the return type of the drivers' routines, there is not much we can do it at a lower level. But I believe that we should not unify at the level of `.to_dict()`, that should have a unique behavior.

Concerning the `platform.sweep(average=True)`, there is an alternative: provide a `platform.sweep_average()` function (in the simplest case just an alias, in the best one implemented with the same ingredients, breaking `sweep()` in more atomic functions and keep the averaging part only in `sweep_average()`).
The advantages would be that:
- from typing point of view, you would have a clear output (rather than a `Union`, uniquely depending on an input flag)
- but this reflects a more relevant principle: one function should perform one task; if you're returning one type of two, most likely your function is doing multiple jobs, so you might consider splitting it

However, the possible split is definitely beyond the scope of this issue, and it mostly involves drivers.

andrea-pasquale, 2023-05-17:
I'm closing this since it was fixed by qiboteam/qibolab#346

closed 2023-05-17 by andrea-pasquale