qiboteam/qibocal #296: update calibrate_qubit_states

Edoardo-Pedicillo, 2023-04-03:
After the merging of the classifiers, I would like to implement them in the runcard. 
I am thinking of modifying the `calibrate_qubit_states` routine in such a way that the runcard will look like in the following example:
```
 -  calibrate_qubit_states:
      nshots: 2048
      classifiers: ["qubit_fit", "nn"]
      save_dir: path/to/results
```
In the final report there will be a plot of the results for each model in the `classifiers` list. What do you think about it @aorgazf  @scarrazza ?

scarrazza, 2023-04-04:
Sounds good to me, please give a try and post the report layout here so we can have further discussion.

Edoardo-Pedicillo, 2023-04-05:
I tried with the `dummy` and this is the report
[image]

scarrazza, 2023-04-05:
Looks good, but please include a single plot with the roc/auc. Please add metrics for training/valodation/test.

Edoardo-Pedicillo, 2023-04-06:
I added what you requested, now the plots of the report look like this
[image]

scarrazza, 2023-04-06:
Please add some titles otherwise is difficult to understand. Could you please align the plots in the last row to match the plots in the second row?

Edoardo-Pedicillo, 2023-04-10:
For the time being the Routine always performs a hyperparameter optimization, considering it is time-consuming, I would add the possibility to dump the model's hyperparameters somewhere in 'Qibocal' (maybe in 'extras') for the first call of the Routine and skip this step in the future calls.
[image]

scarrazza, 2023-04-10:
Yes, that sounds acceptable. Another option is to store the model name and parameters in the platform runcard.

Edoardo-Pedicillo, 2023-04-10:
I have not a strong opinion on that. Your idea sounds good, so the workflow for each model should be: if the hyperparameters are not in the platform an hyperoptimization is performed and the hyperparameters are loaded in the platform runcard, otherwise this step is skipped, then the model is trained, and the parameters are dumped (following qiboteam/qibolab#285).

scarrazza, 2023-04-10:
That's a possibility, please give a try and see if from a practical perspective this looks smoothly.

Edoardo-Pedicillo, 2023-04-13:
The dumping and loading of the hyperparameters seems to work smoothly, we can meet and discuss it. Now I would like to store the model's parameters after the training, and I'm facing some issues choosing the file format, because we are dealing with models belonging to different libraries: `scikit`, 'tf Keras' and the classifier `qubit_fit` is implemented directly in `Qibocal`.  We have two possibility:

- Store each file format in the more convenient way, i.e. dump the scikit models and qubit_fit with skops (the scikit model's persistence is extensively described here https://github.com/qiboteam/qibolab/pull/285#issuecomment-1406162044) and the keras model in h5 format

- Try to find a common format, in this case the ONNX seems the best solution (see [sklearn-onxx](http://onnx.ai/sklearn-onnx/) and [keras2onxx](url)) the main issue is with `qubit_fit`, one way to dump it in the ONNX format is to rewrite it as a scikit custom model and then implement a [custom converter](https://onnx.ai/sklearn-onnx/tutorial_2_new_converter.html), but I am not sure if this will work
Do you have any suggestions on the best approach?

Edoardo-Pedicillo, 2023-04-14:
I tested the routine on hardware (qw5q_gold) and this is the result
[image]

andrea-pasquale, 2023-08-25:
@Edoardo-Pedicillo this can be closed, right?

Edoardo-Pedicillo, 2023-08-25:
yes

closed 2023-08-25 by Edoardo-Pedicillo