qiboteam/qibocal #300: Add possibility to share runcards using `qq-upload`

andrea-pasquale, 2023-04-10:
Currently in order to share platform runcards in qibolab usually we open a PR in qibolab and/or we
update this [table](https://github.com/qiboteam/qrccluster/wiki/QPU-Status). I think that the table is fine as long as it is updated regularly, instead opening PRs could become tricky especially now that library is not very stable. In particular, we need to merge this type of PR soon to main and/or we need to maintain the PR with the runcard updated to include eventual fixes which could be pushed to main.

To simplify the way in which we share runcards we could also use other tools which are already available in qibocal such as `qq-upload`, as @ingoroth suggested. Even though it hasn't been used as much so far `qq-upload` could be a first prototype of shared database in order to share runcards.

In order to download the runcard directly from `qq-upload` we will need to do some modifications since currently only qibocal reports are visible. 

Let me know what you think @scarrazza @stavros11 @igres26 @ingoroth @Edoardo-Pedicillo @aorgazf @maxhant @DavidSarlle

scarrazza, 2023-04-10:
My suggestion is to follow the approach adopted by vp-upload in NNPDF where any folder can be uploaded to the server. If this folder contains a report, it will be rendered by the browser, otherwise a list of files will be visible for download.

scarrazza, 2023-04-10:
BTW, this approach works for sharing runcards "without compromise", however I still believe we need a separate git repository for runcards, which we can store commit history and provide access to the cluster.

scarrazza, 2023-04-10:
There 2 points:
- the possibility to upload folders, containing whatever inside, independently from the report.
- the possibility to retrieve and install the runcard.
The second point is more tricky, we have vp-get in NNPDF for this which downloads and copy the restive files to the proper location. I am fine with this approach for development, but still need a stable repo for official runcards.

andrea-pasquale, 2023-04-10:
Thanks for the feedback @scarrazza.

> * the possibility to retrieve and install the runcard.

In theory, as long as the user as `qibolab` installed from source this should consist only in copying the runcard in `src/qibolab/runcards/` although I understand that in the future It could not be so straightforward

> I am fine with this approach for development, but still need a stable repo for official runcards.

I totally agree with having a repo for official runcards. In this way we could disentangle them from qibolan. At the of the (auto)calibration we could even have a workflow/bot that can open a PR inside this repo so that people can review the new runcard and if it is good enough they can merge it to main.

ingoroth, 2023-04-13:
I fully agree, @scarrazza and @andrea-pasquale, that a more general sharing mechanism with qq_upload is a valuable feature. At some point it is also good to have a common database of runcards that is actually automatically used on the machines (instead of "hardcoding" them into the qibolab repositories).  But for a quick solution we can make something much simpler: The runcards are already part of the report and uploaded.  In principle one can already find them on the qibocal home folder on the cluster.    

Simply adding a link to the runcards (`report_folder/platform.yml` and `report_foldernew_platform.yml`) when the table is build with  https://github.com/qiboteam/qibocal/blob/219646789ea8821d7b4bd72f46c51a020fc04c65/serverscripts/qibocal-index-reports.py
will already make this retrieving more convenient.

andrea-pasquale, 2023-09-26:
I'm closing this given that now all the runcards have been moved to https://github.com/qiboteam/qibolab_platforms_qrc

closed 2023-09-26 by andrea-pasquale