qiboteam/qibocal #301: Global qubits parameter in runcard to action specific parameter

wilkensJ, 2023-04-10:
Would it be possible to specify the `qubits` parameter in the qibocal runcards in each action instead of globally? When using `niGSC` protocols, it could be very useful to specify target qubits in the same runcard.
But I can see that, when characterizing only one qubit with many actions, the global parameter is very useful.
So instead of
https://github.com/qiboteam/qibocal/blob/9a27c90074150cec3106010a16b2c104011ca451/runcards/niGSC.yml#L1-L28

Have something like

```yaml
backend: numpy
format: pickle

actions:
  standardrb:
    qubits: [0]
    nqubits: 5
    depths: [1,3,5,7,10]
    runs: 2
    nshots: 1024
    noise_model: PauliErrorOnUnitary
    noise_params: [0.01, 0.01, 0.01]
  XIdrb:
    qubits: [2]
    nqubits: 5
    depths: [1,2,3,4,5,6,7,8,9,10]
    runs: 5
    nshots: 10
    noise_model: PauliErrorOnX
    noise_params: [0.05, 0.01, 0.01]
  simulfilteredrb:
    qubits: [0,1,2,3,4]
    nqubits: 5
    depths: [1,3,5,7,10]
    runs: 2
    nshots: 1024
    noise_model: PauliErrorOnUnitary
    noise_params: [0.01, 0.01, 0.01]
```

andrea-pasquale, 2023-04-11:
We already had this discussion in the past.
The reason why currently we only have a global parameter `qubits` is that, at least for low level characterization, usually we execute each routine on all the qubits.
Eventually we could have both a global and a local parameter `qubits` and if the local parameter is not provided the code should fall back on the global variable.
I'm not so sure on what the code should do if both a global and a local parameter `qubits` are provided.

I will also like to hear from other people if this feature could be useful to have @aorgazf @DavidSarlle @maxhant @stavros11 @igres26.

stavros11, 2023-04-11:
> I will also like to hear from other people if this feature could be useful to have @aorgazf @DavidSarlle @maxhant @stavros11 @igres26.

Yes, I think it is required to be able to override the global `qubits` parameter in specific routines. For example, for two-qubit gates we need to run each routine for a pair of qubits instead of each individual qubit as it happens now. This problem already appears for qw25q and for now we are just hard-coding the qubits in the routine which is not good.

PR #351 (2023-06-09): Add optional local qubits parameter
Closes #301 and this point
> - [x] qubits argument (see first presentation of new layout)

from #336.

This PR adds a local `qubits` parameter for each routine.
If this parameter is not provided it will be used the global parameter `qubits`, if provided it will overwrite the global parameter.
Probably this is not enough for supporting two qubit gates routines.
We could address this in another PR once we port two qubit gates to the new layout.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decr [... cut]
closed 2023-06-09 by andrea-pasquale