qiboteam/qibocal #302: Make `format` parameter optional in runcards

wilkensJ, 2023-04-10:
Would it be possible to make the `format` parameter in the qibocal runcards optional? The `format` parameter is not needed for `niGSC` actions, and when not given, an error is thrown:

```txt
File "/nfs/users/jadwiga.wilkens/qibocal/src/qibocal/cli/_base.py", line 58, in command
    builder = ActionBuilder(runcard, folder, force)
  File "/nfs/users/jadwiga.wilkens/qibocal/src/qibocal/cli/builders.py", line 161, in __init__
    self.format = self.runcard["format"]
KeyError: 'format'
```

andrea-pasquale, 2023-04-11:
> The `format` parameter is not needed for `niGSC` actions

Currently it is not needed since you hardcoded the format to be `pkl` for `niGSC` actions.
While for circuits you need necessarily to use pickle, for data you could also use `csv`.
I think that a proper solution would be to include also the parameter `format` in `niGSC` actions.

wilkensJ, 2023-04-11:
Thank you for taking on this issue :)

The experiment class and the module functions know how to save themselves (and their content), there is no `format` parameter needed. If there would be another, more efficient way how to save circuits or data, they can be set for each module, meaning change is hardcoded for each module.
Hence always having to put a `format` in the runcard even though it has no meaning for the niGSC modules is misleading.

andrea-pasquale, 2023-04-12:
> The experiment class and the module functions know how to save themselves (and their content), there is no `format` parameter needed.

I would like to reduce the asymmetries between the niGSC and the rest of the code, therefore I believe that you should also offer the possibility to save into different formats other than `pickle` whenever possible.

wilkensJ, 2023-04-12:
Just be to be clear, the parameters `backend` and `platform` are optional in the runcard which I think is very useful because they don't always have to be set.
https://github.com/qiboteam/qibocal/blob/219646789ea8821d7b4bd72f46c51a020fc04c65/src/qibocal/cli/builders.py#L147-L161
Now the same thing applies for the `format` parameter, it would be quite a big change in many functions (I think in 4)  in the `niGSC` part to make use of the `format` parameter, and in my opinion it would add unnecessary complexity and no advantage since right now the only clean way of how to save circuit objects is pickle.  

On the other hand, by just changing the one line in the `ActionBuilder`

https://github.com/qiboteam/qibocal/blob/219646789ea8821d7b4bd72f46c51a020fc04c65/src/qibocal/cli/builders.py#L161
to 
```python
self.format = self.runcard.get("format", None) 
```
the problem would be solved.

ingoroth, 2023-04-13:
The data of niGSC protocols is complex and writing different storage methods is unnecessary just to make use of a parameter in the run card at this point.  I also think that requiring a format parameter in the run card is an unnecessary complications at the moment and Jadwiga's solution is an easy fix.

closed 2023-05-03 by wilkensJ