qiboteam/qibocal #305: Single shot classification wrong for qblox

andrea-pasquale, 2023-04-21:
After some debugging I found out that the `threshold` and the `rotation_angle` computed by the routine [`calibrate_qubit_states` ](https://github.com/qiboteam/qibocal/blob/598e8622c2d9419adc236dd9d28771adaefd5fd9/src/qibocal/calibrations/characterization/calibrate_qubit_states.py#L12)are not the exact parameters used by qblox. Currently in qibocal we are using Ramiro's method which consist in the following:

1. Translate 0s to the origin and 1s accordingly
2. Rotate 1s to x axis
3. Compute the threshold by looking at the cumulative distributions

According to their [documentation](https://qblox-qblox-instruments.readthedocs-hosted.com/en/master/tutorials/conditional_playback.html#Measure-qubit-histogram) qblox are doing the following:

```py
rotation = np.mod(-np.angle(np.mean(state1)-np.mean(state0)), 2*np.pi)
threshold = (np.exp(1j*rotation)*(np.mean(state1)+np.mean(state0))).real/2
```
which I don't fully understand but it is working and it doesn't involve the average of the state 0.
Here is a comparison between the two different methods using random numbers.
[image]

The final results is the same however they have different values for threshold and angle:
|   | ramiro  | qblox  | 
|---|---|---|
| angle  | -0.45497  | 0.45497  |  
| threshold  | 1.230  | -0.655  |   

As a test I tried to compute the angle and the rotation using qblox method and then I ran allXY which are now working. You can see the report [here](http://login.qrccluster.com:9000/qU70t2QkToSw2YU8vtpQUQ==/).

It would be nice to benchmark also this method @Edoardo-Pedicillo.
If it is faster than what we currently have in main we could use this a default since it is required by qblox.
I'm going to update the runcard with the new classification parameters in a new branch in qibolab so that everyone can use it.
Let me know what you think @stavros11 @scarrazza @Edoardo-Pedicillo @aorgazf @DavidSarlle.

aorgazf, 2023-04-22:
Hi Andrea,
In their example, qblox use the middle point between the mean state0 and mean state1 to discriminate states.
The method we had implemented is slightly better than that, we choose the threshold that maximises fidelity.
The parameters needed by qblox to classify states in hardware are:
- rotation_angle: defined as the angle of the rotation that needs to be applied so that the mean of state1 and the mean of state0 have the same imaginary values; and state1 real value is larger than that of state0.
- threshold, the real value that after applying the rotation, better discriminates between states 0 and 1. Mind that this threshold is measured from the origin of coordinates, not from the mean of state0.
[image]

I've raised a PR undoing the changes introduced by @Edoardo-Pedicillo, and I've committed in qibolab the new classification parameters to the runcard in https://github.com/qiboteam/qibolab/pull/375

Edoardo-Pedicillo, 2023-04-25:
It follows the benchmark of the qublox method (called `qublox_fit`) on qubit 1. The performances are similar to `qubit_fit`, obviously the training time is considerably lower.

[image]
[image]

commit 40d4db44 (2023-05-03): Merge pull request #308 from qiboteam/alvaro/fix_classification_parameters

Fix Classification Parameters closes #305
closed 2023-05-03 by andrea-pasquale