qiboteam/qibocal #322: niGSC tests fail with the new Qibo version (0.1.13)

Edoardo-Pedicillo, 2023-05-04:
As the title says, in particular the ones in tests/test_niGSC_xidrb.py, can you fix it asap?

wilkensJ, 2023-05-04:
Thank you  @Edoardo-Pedicillo I will have a look now.

wilkensJ, 2023-05-04:
I think this is due to the noise model being initiated differently, this is responsible for the failed tests I think:

```python
circuit = models.Circuit(1)
circuit.add([gates.X(0),gates.M(0)])

noise = NoiseModel()
noise_pauli = PauliError(0.1, 0.1, 0.1)
noise.add(noise_pauli, gates.X)

noisy_circuit = noise.apply(circuit)
noisy_circuit.execute(nshots=10)
```

which gives an error:

```txt
    noise_pauli = PauliError(0.1, 0.1, 0.1)
TypeError: __init__() takes 2 positional arguments but 4 were given
```

wilkensJ, 2023-05-04:
I cannot seem to make it work. This also gives raise to an error:
```python
circuit = models.Circuit(1)
circuit.add([gates.X(0),gates.M(0)])

noise = NoiseModel()
noise_pauli = PauliError([0.1, 0.1, 0.1])
noise.add(noise_pauli, gates.X)

noisy_circuit = noise.apply(circuit)
noisy_circuit.execute(nshots=10)
```
```txt
  File "playground.py", line 46, in <module>
    noisy_circuit = noise.apply(circuit)
  File "/Users/jadwigawilkens/Physics/TII/qibo_universe/qibo/src/qibo/noise.py", line 300, in apply
    noisy_circuit.add(error.channel(q, error.options))
  File "/Users/jadwigawilkens/Physics/TII/qibo_universe/qibo/src/qibo/gates/channels.py", line 341, in __init__
    for pauli, probability in operators:
TypeError: cannot unpack non-iterable float object
```

@renatomello is also aware of that problem now.

wilkensJ, 2023-05-04:
So apparently the initialization of the `PauliError` channel changed. I am sorry, I did not know that. @vodovozovaliza  and me will fix that.

renatomello, 2023-05-04:
https://github.com/qiboteam/qibo/pull/807

https://github.com/qiboteam/qibo/pull/869

> I cannot seem to make it work. This also gives raise to an error:
> 
> ```python
> circuit = models.Circuit(1)
> circuit.add([gates.X(0),gates.M(0)])
> 
> noise = NoiseModel()
> noise_pauli = PauliError([0.1, 0.1, 0.1])
> noise.add(noise_pauli, gates.X)
> 
> noisy_circuit = noise.apply(circuit)
> noisy_circuit.execute(nshots=10)
> ```
> 
> ```
>   File "playground.py", line 46, in <module>
>     noisy_circuit = noise.apply(circuit)
>   File "/Users/jadwigawilkens/Physics/TII/qibo_universe/qibo/src/qibo/noise.py", line 300, in apply
>     noisy_circuit.add(error.channel(q, error.options))
>   File "/Users/jadwigawilkens/Physics/TII/qibo_universe/qibo/src/qibo/gates/channels.py", line 341, in __init__
>     for pauli, probability in operators:
> TypeError: cannot unpack non-iterable float object
> ```
> 
> @renatomello is also aware of that problem now.

https://github.com/qiboteam/qibo/pull/807

https://github.com/qiboteam/qibo/pull/869

wilkensJ, 2023-05-05:
@Edoardo-Pedicillo here is @vodovozovaliza PR: https://github.com/qiboteam/qibocal/pull/323

closed 2023-05-10 by Edoardo-Pedicillo