qiboteam/qibocal #353: A clearer separation of characterizations and calibrations

wilkensJ, 2023-05-24:
From the presentation in this issue #352. Already mentioned here #336 (first point), here is just a little more details.

More details
------------
If a characterization is run, it should not try to update anything.
If a calibration is run, it should be clear what is updated.

Why
----
A characterization can be run for different reasons: 
-> Calibrating
-> check if a figure of merit is/is not within tolerance
-> in general to gather data,
…

Suggestions
-------------
-> Write a wrapper which takes a characterization and makes a 
calibration out of it, or
-> Add a flag to switch on/off update parameters  in a routine

[image]

andrea-pasquale, 2023-05-24:
Thanks @wilkensJ for opening this issue.
> -> Add a flag to switch on/off update parameters in a routine

I was thinking that to avoid having many arguments for each routine we could have a global option to avoid updating the platform after each routine. I think that it should be flexible enough.

Something like this

```py
qq-auto --no-update <runcard.yml>
```
will give you the option to run experiments without updating the platform.
What do you think?

wilkensJ, 2023-05-24:
Thanks @andrea-pasquale, I thought more about how routines can be used just to check if a routines did its job. For example @igres26 ran standard rb, then Ramsey, then standard rb again, to check if it the ramsey routine did a good job. Of course right now standard rb does not even has a parameter which it wants to update, but this example shows that sometimes some characterization methods are only used to be looked at or even, for the compressive gate set tomography, just to gather the data (and then send it).

Maybe a global parameter can be set, and if nothing is specified in the runcard for the routines they take the global one as default? But it is still possible to set single flags? What do you think about that?