qiboteam/qibocal #364: Update qibolab dependency

scarrazza, 2023-05-26:
We should update qibolab's dependency to the latest pypi release if no further fixes are needed.

andrea-pasquale, 2023-05-29:
I could update it to the latest pypi release. However, as soon as we merge #365 tests will fail because qiboteam/qibolab#438 will be needed. Are you planning to have a new qibolab release after merging qiboteam/qibolab#438?

AleCandido, 2023-09-11:
We do not pay for releases, so I'd release Qibolab instead, even immediately (if there is nothing major, we can make a minor, otherwise a major is fine).

Please, avoid as much as possible to use Git branches, including `master`.
If you really need, keep this isolated in PRs, but switch it to a released version before merging. Ideally, Qibocal should be even developed with Qibolab released versions, and only add new Qibolab features after they are released (in principle, one could start making pre-releases, but given that it's rather unstable as it is, I do not believe formal pre-releases are worth it).

The rationale for this is that too many moving parts make simpler to create complex situations, and using branches makes impossible to reproduce issues.

andrea-pasquale, 2023-09-11:
I know that of course it is better to have releases, but my main worry right now is to avoid having that many PRs opened.
Personally I would prefer to keep `main`, and try to merge as many PRs until the end of the month ideally and then we can start having releasing.

Making a release now is an option, but it will not be sufficient since there is at least another PR https://github.com/qiboteam/qibocal/pull/450 plus the one that will be developed on top of this one to port errors to all protocols.

AleCandido, 2023-09-11:
If you're worried about https://github.com/qiboteam/qibolab/pull/571, we could merge before and after that.

As I said, we can make as many releases as we want. The fewer the better, but only given that:
- releases are always working (tests included)
- they only depend on other releases

In case we need more to satisfy the two conditions above, we should not worry about doing more.
Better to release in the two consecutive days, rather than releasing something broken or not reproducible.

So, for current Qibocal `main` is only working with Qibolab `main`, I'd release Qibolab immediately. As soon as we'll be close to merge #450 (that doesn't seem to be the case), we will release again (but I would only merge #450 the moment the upstream https://github.com/qiboteam/qibolab/pull/571 will be merged).

scarrazza, 2023-09-11:
My point is very practical, we must update the pyproject with something different from qibolab 0.1.0, otherwise we cannot revert the TWPA. Main or another release is fine by me.

andrea-pasquale, 2023-09-11:
To me if you are going to make releases often (and by often I mean every time that there is a breaking change) we can proceed by just making releases. Having `main` depending on `main` is also fine to me.
@scarrazza given that this decision is going to affect both qibocal and qibolab let me know what would like to proceed.

scarrazza, 2023-09-11:
Let's release qibolab asap. Is there something else waiting for merge that we could sort-out today?

andrea-pasquale, 2023-09-11:
I think that probably also the PR with errors could be merged today, right @Edoardo-Pedicillo?

andrea-pasquale, 2023-09-17:
I'm closing this since now qibocal depends correctly on the stable release of qibolab.

closed 2023-09-17 by andrea-pasquale