qiboteam/qibocal #374: Enhance and expand standard RB and its features

wilkensJ, 2023-06-01:
Since the standard randomized benchmarking protocol PR #349 is close to being ready to be merged to main, I would like to collect all the ideas/requests/feedback which came up during the review and go beyond the scope of this PR. Thanks to @Jacfomg and @andrea-pasquale for the feedback, if something is missing here please feel free to add points.

- [ ] Make the global qubits statement in the qibocal runcard accessible to the data acquisition function. For that the `RBParameters` needs to be updated/`qubits` attribute overwritten when None is given for the qubits in the initialization in the acquisition function.
- [ ] Make the nshots be readable from the platform when not specified in the runcard.
- [ ] Estimate SPAM error using A and B from the exponential fit.
- [ ] Add T1 and T2 to the table
- [ ] Write a module using pre transpiled gates and an own compiler to have full control of the pulses. Maybe add unrolling of sequences as an option.

andrea-pasquale, 2023-12-18:
@Jacfomg can we close this?

Jacfomg, 2023-12-18:
I don't think we are estimating SPAM yet, for T1 and T2 table I feel the monitoring tool would make a better job that reporting it there. For the last, I am not aware of the compiler thing they wanted to make, but we do have unrolling.

andrea-pasquale, 2025-02-14:
Closing this since it is currently outdated as an issue.

closed 2025-02-14 by andrea-pasquale