qiboteam/qibocal #377: niGSC tests fail with Qibo 0.1.14

Edoardo-Pedicillo, 2023-06-02:
https://github.com/qiboteam/qibocal/actions/runs/5153461808/jobs/9280721658#step:11:2411

wilkensJ, 2023-06-02:
Thank you @Edoardo-Pedicillo, is the qibo version for qibocal updated to the latest one? Then the tests fail because of the new implementation of the noise models.

Edoardo-Pedicillo, 2023-06-02:
>is the qibo version for qibocal updated to the latest one? 

It will be after the merging of #376 (or another PR that updates `poetry.lock`)

>Then the tests fail because of the new implementation of the noise models.

Can you fix them?

wilkensJ, 2023-06-02:
Can you point me to the PR which updates the version? Then I can push the changes to that branch so that it is synchronized.

Edoardo-Pedicillo, 2023-06-02:
The PR is #376

closed 2023-06-05 by Edoardo-Pedicillo