qiboteam/qibocal #378: Migrate qubit_spectroscopy_flux_w01 and w02 to autocalibration

aorgazf, 2023-06-02:
Hi @DavidSarlle,
Could you please take care of migrating these two routines to autocalibration so that we can automatically extract the sweet spots.

https://github.com/qiboteam/qibocal/tree/alvaro/qblox9_sweepers

andrea-pasquale, 2023-06-02:
I think that this issue has some overlap with #367 which aims at implementing the fitting for flux routines in the new layout. I think that we need to discuss what are the differences between the fitting implementedy by @AlejandroSopena and the fitting implemented in that branch.

DavidSarlle, 2023-06-02:
@aorgazf sure. @andrea-pasquale Meanwhile, I will adapt the routines in order to be compatible with autocalibration. We can modify the fitting (if needed) as soon as we decide about it.

andrea-pasquale, 2023-06-02:
If we plan on adding new routines that are not currently I would prefer if you open first a small PR pointing to main with only those routines, keeping the old layout. I know that this might generates some delay but I  think that this is useful.
Moreover, that branch is outdated that is why I suggesting to proceed as I explained above.

DavidSarlle, 2023-06-02:
@andrea-pasquale  I will open a PR for each new routine that we want to add, starting from the current main. That was my idea. Is that ok for you?

AlejandroSopena, 2023-06-02:
I think it would be better to use the fit that I implemented in PR #263 . This fit allows us to extract various useful parameters from resonator spectroscopy, such as Ec, Ej, asymmetry, readout coupling, a prediction of the qubit frequency. Then, with flux qubit spectroscopy, it is possible to refine some of these parameters and obtain the sweetspots and the elements of the inverse of the flux matrix along with the offset.
I have seen that the fit in this branch https://github.com/qiboteam/qibocal/tree/alvaro/qblox9_sweepers is a parabola, so it only allows extracting an approximation of the sweetspots.
One option is to migrate the flux spectroscopies to the new layout, and then I can update the fit.

andrea-pasquale, 2023-06-02:
Thanks @AlejandroSopena.
The flux spectroscopies are already ported to the new layout. We only need to add the fit.

andrea-pasquale, 2023-06-05:
> @andrea-pasquale I will open a PR for each new routine that we want to add, starting from the current main. That was my idea. Is that ok for you?

Ok @DavidSarlle, from main it is fine.

closed 2023-08-18 by aorgazf