qiboteam/qibocal #395: Save the fitting

Jacfomg, 2023-06-12:
For the #383 I need to save from the fitting part a numpy array on the report. But I would say that this can be useful in general.

AleCandido, 2023-06-12:
The goal has always been that the acquired data and fitting results should be saved on disk. Always :)

So, in this sense, this should be aligned with #372 and its follow-ups.

andrea-pasquale, 2023-06-12:
Great!
I propose `yaml` or `json` given that I'm planning to convert the dataclass `Results` into a dictionary and from there I thought that `yaml` or `json` could be a valuable option.
Since we are refactoring between the two which is more adapt given what we need to do?

andrea-pasquale, 2023-06-13:
Thanks for the detail explanation @AleCandido!

PR #403 (2023-06-28): Storing fitting results
Closes #395 by storing for each protocol the `Results` object as a json.
I had to modify a few things in some of the routines given that `np.ndarray` and `complex` are not json serializable. 
I would like to merge this PR  **after**   #372. 
I'm also planning to open a PR to remove `complex` for `mean_gnd_states` and `mean_exc_states` I think that a list should be more than enough.

Checklist:
- [ ] Reviewers confirm new code works as expected.
- [x] Tests are passing.
- [x] Coverage does not decrease.
- [x] Documentation is updated.
closed 2023-06-28 by andrea-pasquale