qiboteam/qibocal #416: Time of flight and smearing fitting

Jacfomg, 2023-06-26:
I'm still investigating this routine, it seems that it's best to send powerful pulses not at the readout frequency at least for the chip I used.

The fitting is still very tricky becuase of the high frequency oscillations on the signal and the oscillations on the readout pulse. Those need to be detected and fitted for the smearing value (marked red samples). Any advice ?

[image]

andrea-pasquale, 2023-06-26:
If the main problem are the oscillations what about using FFT to remove them?
See for example [here](https://stackoverflow.com/questions/39799821/how-to-remove-frequency-from-signal).

Jacfomg, 2023-06-26:
The problem is that @rodolfocarobene didn't see these oscillations, just some small noise. I still not sure until we test more instruments or devices if it's only happening for 2D resonators

andrea-pasquale, 2024-10-02:
Closing this in favor of #876.

closed 2024-10-02 by andrea-pasquale