qiboteam/qibocal #419: AllXY is not giving proper information

Jacfomg, 2023-06-28:
https://arxiv.org/abs/2106.06173

If we want to follow their graphs to extract information about using ALLXY you can see on figure.11 how the have 42 points out of the 21 combination of gates. That is because you also need to run both pairs [Ex. (I-pi_pulse) and (pi_pulse-I)] with what I assume would be the identity having some wait time associated to it.

andrea-pasquale, 2023-06-29:
Thanks @Jacfomg. We can give it a try.
I think that the identity should have a default wait time in qibolab, right?

Jacfomg, 2023-06-29:
I was thinking of the same as the pi_pulse as all the gates have that duration on that routine.

Jacfomg, 2023-07-03:
Okay, it's properly explained here, I will try have a look to it soon.
https://arxiv.org/pdf/1311.6759.pdf

andrea-pasquale, 2023-07-03:
I was also looking at this reference earlier. However, I don't seems to find the part where they run both pairs.
Also the drag pulse fitting is very interesting and it does not seems so complicated to implement (Fig.5.8b).

Jacfomg, 2023-07-04:
I had a look and I agree, that fitting seems very interesting. Also, Fig 5.6 having values for the detuning and pi-pulse power that may allow some sort of fitting ? For the pairs it just seems like they run the same 21 sequences we have twice.

Jacfomg, 2023-07-04:
Btw, (Fig.5.8b) should be the drag_pulse_tuning

andrea-pasquale, 2023-07-04:
I know, funny enough I believe that **it is not** what we are doing...

Jacfomg, 2023-07-04:
So, I was thinking of changing the AllXY at least to change the x_axis for the sequence instead of the number. But, for duplicating the sequences twice I am not sure if there is any advantage, should we just mimic the paper or not ?

andrea-pasquale, 2023-07-04:
I also don't know about duplicating the sequences. The purpose of duplicating them is just to add statistics? If so I believe that it is not necessary since we are planning to add error bars for all the routines.

Jacfomg, 2023-07-05:
Okay, they it also looked a bit strange I suppose the point was to ease the corrections they mentioned on the paper.

Jacfomg, 2023-07-06:
> I know, funny enough I believe that **it is not** what we are doing...

It doesn't run for me, I get a `ndarray is not JSON serializable`.

closed 2023-07-10 by Jacfomg