qiboteam/qibocal #420: Quantum non-demolition-ness of the readout pulse

Jacfomg, 2023-06-30:
We need a way of checking the quantum non-demolition-ness of our readout pulse before trying to play with #399. I propose trying to implement the scheme presented https://arxiv.org/abs/2106.06173.

andrea-pasquale, 2023-07-03:
Sounds good to me. Are you planning to implement this experiment, right?
[image]
It could also be nice to compute the fidelity which should be equal to the one computed from the classification experiment.

Jacfomg, 2023-07-03:
Yes, that would be the first idea when I get some spare time.

Jacfomg, 2023-07-04:
Not mine at least, but the devices do.

andrea-pasquale, 2023-07-04:
> Not mine at least, but the devices do.

Is this available for qm, qblox or the rfsoc? @rodolfocarobene @stavros11 @aorgazf

Jacfomg, 2023-07-05:
Working for some new routine with is a quantum Zeno prototype. I will try to finish the routine and open PR for the Zurich fix and that routine first.
[image]

Jacfomg, 2023-07-05:
Btw, this is doing the same with 1 shot. You can see readout pulse 40 or around that excited the qubit. I don't know up to what extend this is expected.

[image]

Jacfomg, 2023-07-07:
It's destructive :)

[image]

closed 2023-07-10 by Jacfomg