qiboteam/qibocal #457: Dependencies issues (August 2023)

aorgazf, 2023-08-09:
Hi,
I have created a fresh environment (python==3.9) installed qibocal and qibolab using poetry, 
```bash
conda create -n august python=3.9
conda activate august

cd qibocal
git checkout main
git pull
poetry install --with test

cd ../qibolab
git checkout main
git pull
poetry install --all-extras
```
The installation completed without errors or warnings, however, when trying to run autocalibration I encountered the following errors:
- ImportError: Numba needs NumPy 1.24 or less
```
Traceback (most recent call last):
  File "/home/users/alvaro.orgaz/.conda/envs/august/bin/qq", line 3, in <module>
    from qibocal import command
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/__init__.py", line 5, in <module>
    from .cli import command, upload
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/__init__.py", line 2, in <module>
    from ._base import command, upload
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/_base.py", line 14, in <module>
    from ..cli.builders import ActionBuilder
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 9, in <module>
    from qibocal.auto.execute import Executor, History
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/execute.py", line 10, in <module>
    from .graph import Graph
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/graph.py", line 5, in <module>
    from .task import Task
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/task.py", line 8, in <module>
    from ..protocols.characterization import Operation
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/protocols/characterization/__init__.py", line 5, in <module>
    from .allxy.drag_pulse_tuning import drag_pulse_tuning
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/protocols/characterization/allxy/drag_pulse_tuning.py", line 17, in <module>
    from ..utils import V_TO_UV
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/protocols/characterization/utils.py", line 6, in <module>
    from numba import njit
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/numba/__init__.py", line 55, in <module>
    _ensure_critical_deps()
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/numba/__init__.py", line 42, in _ensure_critical_deps
    raise ImportError("Numba needs NumPy 1.24 or less")
ImportError: Numba needs NumPy 1.24 or less
```
It seems the installation of qibocal installs numpy==1.24.3, but the installation of qibolab upgrades it to numpy==1.25.2.
I have manually downgraded it to 1.24.3 with 
```
pip install numpy==1.24.3
```
and that seems to fix the issue.

On the next attempt I get the following error:
- ModuleNotFoundError: No module named 'pyrsistent'
```
[33;20m[Qibocal 0.0.3|WARNING|2023-08-09 11:31:20]: Deleting previous directory /home/users/alvaro.orgaz/reports/20230809_qw5q_gold_auto.[0m
[38;20m[Qibocal 0.0.3|INFO|2023-08-09 11:31:20]: Creating directory /home/users/alvaro.orgaz/reports/20230809_qw5q_gold_auto.[0m
Traceback (most recent call last):
  File "/home/users/alvaro.orgaz/.conda/envs/august/bin/qq", line 6, in <module>
    sys.exit(command())
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/_base.py", line 58, in command
    builder = ActionBuilder(card, folder, force, update=update)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 38, in __init__
    self.meta = self._prepare_output(runcard)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 75, in _prepare_output
    if self.backend.name == "qibolab":
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 56, in backend
    return self.runcard.backend_obj
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/functools.py", line 969, in __get__
    val = self.func(instance)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/runcard.py", line 58, in backend_obj
    GlobalBackend.set_backend(self.backend, self.platform)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qibo/backends/__init__.py", line 84, in set_backend
    cls._instance = construct_backend(backend, platform, runcard)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qibo/backends/__init__.py", line 35, in construct_backend
    return QibolabBackend(platform, runcard)
  File "/nfs/users/alvaro.orgaz/qibolab/src/qibolab/backends.py", line 24, in __init__
    self.platform = create_platform(platform, runcard)
  File "/nfs/users/alvaro.orgaz/qibolab/src/qibolab/__init__.py", line 52, in create_platform
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/users/alvaro.orgaz/qibolab_platforms_qrc/qw5q_gold_qblox.py", line 6, in <module>
    from qibolab.instruments.qblox.cluster import (
  File "/nfs/users/alvaro.orgaz/qibolab/src/qibolab/instruments/qblox/cluster.py", line 40, in <module>
    from qblox_instruments.qcodes_drivers.cluster import Cluster as QbloxCluster
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qblox_instruments/__init__.py", line 6, in <module>
    from qblox_instruments.qcodes_drivers import Pulsar, Cluster, SpiRack
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qblox_instruments/qcodes_drivers/__init__.py", line 1, in <module>
    from qblox_instruments.qcodes_drivers.pulsar import Pulsar
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qblox_instruments/qcodes_drivers/pulsar.py", line 11, in <module>
    from qcodes import Instrument, InstrumentChannel, Parameter
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qcodes/__init__.py", line 10, in <module>
    import qcodes.configuration as qcconfig
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qcodes/configuration/__init__.py", line 1, in <module>
    from .config import Config, DotDict, logger
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/qcodes/configuration/config.py", line 12, in <module>
    import jsonschema
  File "/home/users/alvaro.orgaz/.local/lib/python3.9/site-packages/jsonschema/__init__.py", line 14, in <module>
    from jsonschema._types import TypeChecker
  File "/home/users/alvaro.orgaz/.local/lib/python3.9/site-packages/jsonschema/_types.py", line 6, in <module>
    from pyrsistent import pmap
ModuleNotFoundError: No module named 'pyrsistent'
```
it seems this dependency is missing in the toml file
`pip install pyrsistent` fixed it

aorgazf, 2023-09-23:
Hi @Edoardo-Pedicillo ,
On my last attempt, pyrsistent issue seems to be fixed, but not the `pip install numpy==1.24.3`

andrea-pasquale, 2023-09-27:
> Hi @Edoardo-Pedicillo ,
> On my last attempt, pyrsistent issue seems to be fixed, but not the `pip install numpy==1.24.3`

This behavior is partially expected since qibocal depends on `numpy==1.24`
https://github.com/qiboteam/qibocal/blob/46346a104c07f5ff74fa4076e239a9dcc77983e7/pyproject.toml#L22
while qibolab does not have a specific requirement for numpy therefore any version can be installed. 

A possible solution would be to just fix also for qibolab `numpy==1.24`

closed 2023-10-23 by andrea-pasquale