qiboteam/qibocal #458: 'Platform' object has no attribute 'dump' Error

aorgazf, 2023-08-09:
Hi,
Attempting to run autocalibration on qw5q_gold (connected to qblox), using main in all repositories, I get the following error:
- AttributeError: 'Platform' object has no attribute 'dump'
```
[33;20m[Qibocal 0.0.3|WARNING|2023-08-09 11:40:42]: Deleting previous directory /home/users/alvaro.orgaz/reports/20230809_qw5q_gold_auto.[0m
[38;20m[Qibocal 0.0.3|INFO|2023-08-09 11:40:42]: Creating directory /home/users/alvaro.orgaz/reports/20230809_qw5q_gold_auto.[0m
[Qibo 0.1.16|INFO|2023-08-09 11:40:43]: Loading platform qw5q_gold_qblox
[Qibo 0.1.16|INFO|2023-08-09 11:40:43]: Using qibolab (qw5q_gold_qblox) backend on /CPU:0
Traceback (most recent call last):
  File "/home/users/alvaro.orgaz/.conda/envs/august/bin/qq", line 6, in <module>
    sys.exit(command())
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/users/alvaro.orgaz/.conda/envs/august/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/_base.py", line 58, in command
    builder = ActionBuilder(card, folder, force, update=update)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 38, in __init__
    self.meta = self._prepare_output(runcard)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/builders.py", line 76, in _prepare_output
    self.platform.dump(self.folder / PLATFORM)
AttributeError: 'Platform' object has no attribute 'dump'
```
@DavidSarlle has been able to reproduce the same error.

aorgazf, 2023-08-09:
The same error happens with tii1q_b1 connected to rfsoc4x2

stavros11, 2023-08-09:
`Platform.dump` was removed from qibolab yesterday. The fix for qibocal is in #456.

aorgazf, 2023-08-09:
Thanks Stavros.
I have provided my approval to the PR and marked it as urgent. At the moment main is broken.
Hopefully we can have it merged asap.

closed 2023-08-09 by aorgazf