qiboteam/qibocal #463: 2q gates - Cryoscope

aorgazf, 2023-08-15:
Hi,
Up until now we have managed to apply some basic corrections to the flux pulses that implement 2q-gates.
The manual process is slow and does not render the best correction. In order to get better 2q-gates we need to be able to run the cryoscope routine.
There have been several attempts at writing this routine in the past, but at that time we could not use RT sweepers, so the progress was slow.
In the last implementation, I believe the acquisition of the data and its processing was correct up until the end, when the IIR and FIR filter factors needed to be calculated.
That routine was written for the previous version of qibocal and it now needs to be ported and fix the bug in the calculation of the coefficients of the filters.

aorgazf, 2023-08-18:
Dear @stavros11,
I'm not sure how much time @rodolfocarobene  is going to be able to dedicate to this issue.
Considering its importance, it would be nice if you could work on it.

aorgazf, 2024-06-20:
Hi,
I have been working on this routine and I think I have a working version in a notebook.
[20240528_iqm5q_cryoscope_0.1_all_qubits_padded.zip](https://github.com/user-attachments/files/15917063/20240528_iqm5q_cryoscope_0.1_all_qubits_padded.zip)
the acquisition code is here:
src/qibocal/protocols/characterization/z/in_progress/cryoscope.py
and the fit is in this notebook
[cryoscope fit.zip](https://github.com/user-attachments/files/15917120/cryoscope.fit.zip)

andrea-pasquale, 2024-06-21:
Hi @aorgazf, thanks for sharing this.
Given that many of us tried before, if your code is working feel free to open a PR so that we can have your modifications in main.

closed 2025-03-13 by andrea-pasquale