qiboteam/qibocal #467: Resonator and Qubit Flux Spectroscopy Matrices

aorgazf, 2023-08-18:
Hi @DavidSarlle ,
It would be nice to extend the functionality of the flux spectroscopy routines to not only show the elements of the diagonal (how the frequency of a resonator/qubit changes with the flux applied to that qubit) but the entire matrix. Please note that the fitting should not be applied to those off-diagonal elements.
This matrices will allow us to see the crosstalk between fluxlines and the avoided crossings between qubits. These avoided crossings can later be used,in conjunction with the chevron, to characterise 2q gates.

This functionality was implemented in my branch [src/qibocal/calibrations/characterization](https://github.com/qiboteam/qibocal/tree/alvaro/qblox9_sweepers), but never ported to the latest format of qibocal.

andrea-pasquale, 2023-10-20:
Closed by #470

closed 2023-10-20 by andrea-pasquale

stavros11, 2023-10-20:
@andrea-pasquale regarding these issues, I believe we are still missing the fitting, ie the numerical calculation of the crosstalk matrix (not just the plots), but as we discussed in the other PR this may not be easy given that avoided crossings may appear.

andrea-pasquale, 2023-10-20:
Yes, I know that we are missing the fitting. I was going through issues and I decided to close this one given that the major problem was having a protocol which could show how each qubit is affected by moving the flux of another qubit.
Eventually we can open another issue for just the crosstalk matrix.