qiboteam/qibocal #476: Discrepancy between T2 routine with RT sweepers and using for lops

aorgazf, 2023-08-18:
Hi, 
I have noticed a discrepancy between the results obtained running `t2` and `t2_sequences` on qw5q_gold_qblox:

[image]

[image]

The problem may be in the driver.

andrea-pasquale, 2023-08-24:
I think that this might be related to the implementation of the sweeper in qblox.
With @igres26 we observed a similar behavior. It seems that the readout pulse at the end it is not pushed correctly.

andrea-pasquale, 2023-11-21:
Closing this as duplicate of https://github.com/qiboteam/qibolab/issues/650 (first bullet points).

closed 2023-11-21 by andrea-pasquale