qiboteam/qibocal #488: Cannot import name 'autocalibration' from 'qibocal'

JavierSerranoGarcia, 2023-08-25:
I am finding this issue trying to run an experiment. I have pulled the last version of qibocal and used main branch.

_(venv) javier.serrano@saadiyat:~/qiboDev/qubitCar$ sbatch -p qw25q qibocal.sh
Submitted batch job 1021195
(venv) javier.serrano@saadiyat:~/qiboDev/qubitCar$ cat log.log
/nfs/users/javier.serrano/qiboDev/venv/lib/python3.9/site-packages/numba/core/typed_passes.py:334: NumbaPerformanceWarning: 
The keyword argument 'parallel=True' was specified but no transformation for parallel execution was possible.

To find out why, try turning on parallel diagnostics, see https://numba.readthedocs.io/en/stable/user/parallel.html#diagnostics for help.

File "../qibocal/src/qibocal/protocols/characterization/utils.py", line 175:
@njit(["float64[:] (float64[:], float64[:])"], parallel=True, cache=True)
def cumulative(input_data, points):
^

  warnings.warn(errors.NumbaPerformanceWarning(msg,
Traceback (most recent call last):
  File "/nfs/users/javier.serrano/qiboDev/venv/bin/qq-auto", line 5, in <module>
    from qibocal import autocalibration
ImportError: cannot import name 'autocalibration' from 'qibocal' (/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/__init__.py)_

andrea-pasquale, 2023-08-25:
Thanks @JavierSerranoGarcia for opening the issue.

The first part is just a warning, so it should not cause any issues.

> File "/nfs/users/javier.serrano/qiboDev/venv/bin/qq-auto", line 5, in
> from qibocal import autocalibration
> ImportError: cannot import name 'autocalibration' from 'qibocal' (/nfs/users/javier.serrano/qiboDev/qibocal/src/qibocal/**init**.py)_

This is quite weird. Which command are you using in `qibocal.sh`.
`qq-auto` in now deprecated, you should use `qq`.

If that doesn't fix it you can try to uninstall and reinstall qibocal.

JavierSerranoGarcia, 2023-08-25:
Thanks @andrea-pasquale. It was a problem with qq

closed 2023-08-25 by JavierSerranoGarcia