qiboteam/qibocal #514: Stable documentation not generated/rendered

scarrazza, 2023-09-10:
The latest stable documentation is not rendered properly (we still have v0.0.1).

AleCandido, 2023-09-11:
As usual, the workflow seems fine

https://github.com/qiboteam/qibocal/actions/runs/6123719651

The pages artifact is expired, but the other one (the rendered one) is the one related to the tag. So, it has been properly generated, and I'm pretty sure it is also properly included in the whole bundle (even though this time I can't check).

Still no idea about why is not served by the website...

(It's also a bit stranger than usual, since `latest` has been redeployed, but `stable` is still outdated)

scarrazza, 2023-09-11:
Please restart the workflow. I have checked yesterday and the artifact was wrong too.

AleCandido, 2023-09-11:
I restarted the one on the tag. At least I'll be able to inspect the artifact.

AleCandido, 2023-09-11:
I confirm that the `stable` version generated in the `github-pages` artifact is the correct one (and the workflow is the exact same, I just reran).

AleCandido, 2023-09-11:
Btw, now it's also the version served by GitHub is correct, so we may close this issue (this time, why it's occurring is still unknown to me).

scarrazza, 2023-09-11:
Very, very, very strange...

closed 2023-09-11 by scarrazza