qiboteam/qibocal #515: Deploy workflow too long

scarrazza, 2023-09-10:
The deploy workflow for this repo is taking the same amount of time of tests, i.e. we are including also tests. In order to match the other modules, we should disable tests for this specific workflow.

Edoardo-Pedicillo, 2023-09-11:
I could delete the test job from https://github.com/qiboteam/workflows/blob/main/.github/workflows/deploy-pip-poetry.yml, i.e., not test during deployment for all the repositories, or I could add a flag and disable the test job for Qibocal. Which solution do you prefer ?

scarrazza, 2023-09-11:
We should do the same as other repositories.

andrea-pasquale, 2023-09-12:
Btw, the duration of the workflow should be fixed after merging #450.
In any case, if running tests is not totally necessary to deploy the doc we can remove it.

AleCandido, 2023-09-12:
We are attempting to deploy as a packaging test, but not actually deploying. In that case, testing is just doubling the other workflow.

Moreover, we are actually deploying only on tags, and we tag only `main`/`master`, after we merge PRs that have already been tested and passing. So, even for actual deployments, testing right before is quite redundant.

I perfectly agree with the simplification.

closed 2023-09-12 by scarrazza