qiboteam/qibocal #587: Couplers Breaking Changes

aorgazf, 2023-10-26:
Dear all,
`chevron` routine throws the error shown below when trying to run it on qw5q_gold. 
It appears the changes introduced by @Jacfomg to support QPU architectures with couplers were not tested on platforms that do not have couplers.

```
Traceback (most recent call last):
  File "/nfs/users/alvaro.orgaz/.envs/november/bin/qq", line 6, in <module>
    sys.exit(command())
  File "/nfs/users/alvaro.orgaz/.envs/november/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/alvaro.orgaz/.envs/november/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/alvaro.orgaz/.envs/november/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/alvaro.orgaz/.envs/november/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/alvaro.orgaz/.envs/november/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/_base.py", line 53, in auto
    autocalibrate(card, folder, force, update)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/cli/autocalibration.py", line 61, in autocalibrate
    for task_uid in executor.run(mode=ExecutionMode.autocalibration):
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/execute.py", line 136, in run
    completed = task.run(
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/task.py", line 146, in run
    completed.data, completed.data_time = operation.acquisition(
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/auto/operation.py", line 41, in wrapper
    out = func(*args, **kwds)
  File "/nfs/users/alvaro.orgaz/qibocal/src/qibocal/protocols/characterization/two_qubit_interaction/chevron.py", line 128, in _aquisition
    cz.coupler_pulses(platform.pairs[tuple(sorted(ordered_pair))].coupler.name)
AttributeError: 'NoneType' object has no attribute 'name'
```

Jacfomg, 2023-10-26:
Already fixed on https://github.com/qiboteam/qibolab/pull/623, it's waiting for review to merge.

andrea-pasquale, 2023-10-26:
> Already fixed on [qiboteam/qibolab#623](https://github.com/qiboteam/qibolab/pull/623), it's waiting for review to merge.

I tried this branch and it still does not work on `qw5q_gold`

closed 2023-10-27 by andrea-pasquale