qiboteam/qibocal #593: QPU performance bot

scarrazza, 2023-10-27:
We need a performance measurement tool which will extract parameters after execution on QPU for a given platform runcard. This program should store at least T1, T2 and fidelities.

andrea-pasquale, 2023-10-27:
Actually @Jacfomg already coded something similar for the selfhosted in https://github.com/qiboteam/qibolab_platforms_qrc/pull/73
From the qibocal this is achieved by using a specific action runcard, which is then executed using `qq auto`.
From a practical perspective should we just include this runcard here or do you want something more @scarrazza ?

scarrazza, 2023-10-27:
Yes, this works, please add the runcard here.

Jacfomg, 2023-10-27:
This will work for T1, T2 and assigment fidelity. For the RB I still need to fix a couple things on the routine and find some nice parameters to get a fidelity with a short scan.

andrea-pasquale, 2023-10-27:
> Yes, this works, please add the runcard here.

Concerning the runcard I need to specify already the platform and the qubits...
If we want something which is platform agnostic we may want to add the possibility to pass the platform and the qubits directly from the cli (instead of putting them in the runcard).
Something like:

```sh
qq auto <runcard.yml> --platform qw5q_gold --qubits [0,1,2,3] -o <output_folder>
```
Let me know what you think

andrea-pasquale, 2023-10-27:
@stavros11 this could also be useful to simplify the selfhosted workflow I guess

scarrazza, 2023-10-27:
Alternatively, we could have a `qq monitor` which automatically uses your predefined runcard and delegates to qibolab the decision of the platform name.

andrea-pasquale, 2023-10-27:
> Alternatively, we could have a `qq qpu-monitor` which automatically uses your predefined runcard and delegates to qibolab the decision of the platform name.

I see. In order to change the platform this program should be able to edit `yaml` files or dictionaries. Or we need to put a placeholder for the platform

scarrazza, 2023-10-29:
@andrea-pasquale please do whatever is simpler and faster, from my side, I have just coded a strategy to display the information starting from the output of qq.

PR #595 (2023-11-03): Add platform and backend entries to qq commands
Closes #593.
This PR adds the possibility to specify the platform name and the backend directly from command line:

```sh
qq auto <runcard> --platform dummy --backend qibolab
```
Currently if those values are not specified from CLI they will be read from the runcard if they are present.
If they are not specified also in the runcard the default values wil be "qibolab" for backend and "dummy" for platform.

I still need to update documentation but in theory this should be ready in terms of implementation.

Checklist:
- [ ] Reviewers confirm new cod [... cut]
closed 2023-11-03 by andrea-pasquale