qiboteam/qibocal #598: allXY still not working in iqm

DavidSarlle, 2023-11-01:
allXY is still not working in iqm. After the fix introduced by @andrea-pasquale now it looks like:

[image]

Seems that no results for all the pais of gates are shown anymore...

DavidSarlle, 2023-11-02:
@Jacfomg or It is still a driver problem or It is a qibocal bug introduced in latests changes. Why are you assigning the issue too me?

andrea-pasquale, 2023-11-02:
I was debugging the AllXY with @stavros11 and @Jacfomg a week ago. I pushed some changes single the code was not working with qlbox which broke zh driver. I believe that this is mostly a qibolab issue. We should make sure that all drivers return exactly the same type of output (mostly the shape of the array) for all the different execution and acquisition mode that we have in qibolab.

Edoardo-Pedicillo, 2023-11-22:
Closed because https://github.com/qiboteam/qibolab/issues/647

closed 2023-11-22 by Edoardo-Pedicillo