qiboteam/qibocal #601: syntax in results.json

scarrazza, 2023-11-01:
When parsing the output of routines, we generate the results json file with content similar to this:
```
{"\"t1\"": {"0": 21978.285056907527}, "\"fitted_parameters\"": {"0": [148379.5944142615, -13688.143615525545, 21978.285056907527]}}
```
where keys are surrounded by quotation marks, making the parsing more complicated than usual.

andrea-pasquale, 2023-11-01:
Thanks @scarrazza for opening the issue.
I will have a look.

andrea-pasquale, 2023-11-02:
In qibocal we provide a way to deserialize directly  those files in a more user friendly format through the function [`deserialize`](https://github.com/qiboteam/qibocal/blob/3411611b77aa107cfa64a6c6da2f86aacf6fae55/src/qibocal/auto/serialize.py#L4). Here is an example starting from this type of dict:

```sh
{'"t1"': {'0': 3752193.1747832014,
  '1': 6130.3014814689905,
  '2': 17813.4245455305,
  '3': 2614508.9812356164,
  '4': 10002.822084029935},
 '"fitted_parameters"': {'0': [-158659064.0229236,
   -233212652.30242014,
   3752193.1747832014],
  '1': [65150829.57531139, -36523202.42633729, 6130.3014814689905],
  '2': [77064282.40396605, -10233085.468842123, 17813.4245455305],
  '3': [-171867209.9229064, -251465207.57537806, 2614508.9812356164],
  '4': [81507646.2552062, -21115058.520144194, 10002.822084029935]}}
```
After applying deserialize we can get the following dict:

```py
from qibocal.auto.serialize import deserialize
new_data = deserialize(data)
```
```sh
{'t1': {0: 3752193.1747832014,
  1: 6130.3014814689905,
  2: 17813.4245455305,
  3: 2614508.9812356164,
  4: 10002.822084029935},
 'fitted_parameters': {0: [-158659064.0229236,
   -233212652.30242014,
   3752193.1747832014],
  1: [65150829.57531139, -36523202.42633729, 6130.3014814689905],
  2: [77064282.40396605, -10233085.468842123, 17813.4245455305],
  3: [-171867209.9229064, -251465207.57537806, 2614508.9812356164],
  4: [81507646.2552062, -21115058.520144194, 10002.822084029935]}}

```
@scarrazza let me know if this fixes your issue

scarrazza, 2023-11-02:
What is data in your example?

andrea-pasquale, 2023-11-02:
Assuming that your output folder for qq is `path` it should be this (for t1):
```py
data = json.loads((path/ "data/t1_0/results.json").read_text())
```
So it is just the raw data that you load from json.

scarrazza, 2023-11-02:
This works, however I don't understand why this serialization is adding extra quotation marks. Is that a bug or a feature?

andrea-pasquale, 2023-11-02:
It is a bug. In fact I have a TODO
https://github.com/qiboteam/qibocal/blob/158e7e63181c5201085a02624501de33ed6ebed1/src/qibocal/auto/serialize.py#L7
I don't remember exactly why I opted for this solution but I think it was related to the fact that we are also using strings to name qubits.
We can keep the issue open to remember to fix it.