qiboteam/qibocal #618: `qq upload` improvements

GabrielePalazzo, 2023-11-10:
After a brief discussion with @andrea-pasquale and @stavros11 we came up with three ideas/improvements (sorted by priority).
- We should use relative paths for reports, so that users in the lab (therefore those who cannot access login.qrccluster.com) can use links.
- It would be nice to able to run `qq upload` directly from slurm scripts on the cluster.
- It would be nice to have a search field for routine ids, in order to be able to search for experiment reports regardless how they are named.

andrea-pasquale, 2023-11-13:
Thanks @GabrielePalazzo.
@scarrazza if you agree we can start working on the first two points.

> * We should use relative paths for reports, so that users in the lab (therefore those who cannot access login.qrccluster.com) can use links.

This one should be easy to fix.

> * It would be nice to able to run `qq upload` directly from slurm scripts on the cluster.

For this I am not sure. Perhaps it would be a possibility to skip the ssh authentication which is required for uploading reports.

scarrazza, 2023-11-13:
In overall I agree.

> For this I am not sure. Perhaps it would be a possibility to skip the ssh authentication which is required for uploading reports.

qq upload could simply copy the folder to the current server host/data location.

andrea-pasquale, 2023-11-13:
> qq upload could simply copy the folder to the current server host/data location.

Yes, the problem is that now if you try to use `qq upload` by sending a job it will fail because of the ssh authentication.
In order to make people use it more we thought that it would be useful to include it directly in the script that launches the job.