qiboteam/qibocal #631: Calibration issues

Edoardo-Pedicillo, 2023-11-15:
This issue collects some bugs encountered while calibrating:

- [ ] Rabi fit unstable: the curve starts to increase slowly and the fit is not able to handle this.

[image]

 - [ ] qq fit not working:
```
qq[Qibo 0.2.2|ERROR|2023-11-15 16:27:41]: Platform IQM5q does not exist.
Traceback (most recent call last):
  File "/nfs/users/edoardo.pedicillo/routine/bin/qq", line 8, in <module>
    sys.exit(command())
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/nfs/users/edoardo.pedicillo/qibocal/src/qibocal/cli/_base.py", line 150, in fit
    fitting(folder, update)
  File "/nfs/users/edoardo.pedicillo/qibocal/src/qibocal/cli/fit.py", line 29, in fit
    GlobalBackend.set_backend(backend=meta["backend"], platform=meta["platform"])
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/qibo/backends/__init__.py", line 84, in set_backend
    cls._instance = construct_backend(backend, platform, runcard)
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/qibo/backends/__init__.py", line 35, in construct_backend
    return QibolabBackend(platform, runcard)
  File "/nfs/users/edoardo.pedicillo/qibolab/src/qibolab/backends.py", line 22, in __init__
    self.platform = create_platform(platform, runcard)
  File "/nfs/users/edoardo.pedicillo/qibolab/src/qibolab/__init__.py", line 48, in create_platform
    raise_error(ValueError, f"Platform {name} does not exist.")
  File "/nfs/users/edoardo.pedicillo/routine/lib/python3.10/site-packages/qibo/config.py", line 48, in raise_error
    raise exception(message)
ValueError: Platform IQM5q does not exist.

```
in the `meta.json` we should save `iqm5q`.

- [ ] `qq report` not working: after running `qq fit` successfully, the report does not display any plot (tested only with Rabi).

Edoardo-Pedicillo, 2023-11-22:
> - [ ] Rabi fit unstable: the curve starts to increase slowly and the fit is not able to handle this.

This issue seems related to the status of the hardware.

AleCandido, 2023-11-22:
In principle, we could try to raise better issues (more specific than `ValueError`) for these cases. And once we have, we could catch them in the CLI, and give a better message (the full traceback is not that useful, most likely)

andrea-pasquale, 2023-12-18:
Fixed by #651 #659

closed 2023-12-18 by andrea-pasquale