qiboteam/qibocal #633: time_of_flight_readout not working with qblox drivers

aorgazf, 2023-11-16:
Hi,
I'm not sure if the problem is with the drivers or with qibocal routine, but I've tried running this routine with qblox and it does not work.
@PiergiorgioButtarini could you please check whether qblox driver supports these execution parameters:
```
acquisition_type=AcquisitionType.RAW,
averaging_mode=AveragingMode.CYCLIC,
```

PiergiorgioButtarini, 2023-11-16:
I think as of now Qblox doesn't support `AcquisitionType.RAW`. This believe is motivated both from the fact that there is no reference to this type of acquisition in qblox instruments, and also because the modules were always set with `hardware_demodulation == True`.

andrea-pasquale, 2023-11-16:
Thanks @PiergiorgioButtarini for checking. Could you please move this issue to `qibolab`? If a routine is not working for a driver related problem the issue should be opened in qibolab and not here.

PiergiorgioButtarini, 2023-11-16:
Sure, I will open an issue there.

andrea-pasquale, 2023-11-20:
Closing this as duplicate of https://github.com/qiboteam/qibolab/issues/663

closed 2023-11-20 by andrea-pasquale